mcci-catena/arduino-lmic #352: Do not set region frequency by default in lmic_project_settings.h

lnlp, 2019-07-19:
Currently in lmic_project_settings.h the following is defined:
`#define CFG_us915 1`

This is error prone because this setting is hidden from the user and works for users in US only.
New users are not aware and frequent users of the library can easily forget to select the frequency for their region.
When the user's region is not US this means that without modifying the library code the program will compile and run but fail to communication with the TTN network without any warning or meaningful clue to the user. 
Therefore this setting should not be made by default.

Instead make sure that every user explicitly has to select a frequency specific for their region and give a compiler error if they have not done so yet.

lmic_project_config.h defines project scoped settings at library sourcecode level instead of project level. Changing settings in lmic_project_config.h requires modification of library sourcecode which is maintenance intensive, error prone and difficult to automate.
Maybe there is is a more elegant way to implement this?

paulalting, 2019-07-21:
This is a very good point made by @lnlp , in the case of multiple projects needing specific settings for the library. I found this to be the case myself as I use Eclipse IDE and had to make project based configuration settings for the LMIC library.

I agree, many new users may well not understand how to or forget to set these settings.

terrillmoore, 2019-07-31:
Thanks for your input.

I understand your frustration, but the root cause is a problem with the Arduino environment, aggravated by inadequate documentation in this project. Users would like a way to specify this without editing a file, and either you have to change `boards.txt` or do a LOT of work to make the bandplan selection happen at link time rather than compile time. This would not be acceptable to our 8-bit users.

This issue has been a problem since the beginning of the Arduino LMIC code, except that the original LMIC used EU by default. The solution, if you're using Eclipse or platformio, is to set the undocumented flag `ARDUINO_LMIC_PROJECT_CONFIG_H` to point to a different (or empty) config file.

See:

https://github.com/mcci-catena/arduino-lmic/blob/f75ae27a0964c164a7275555d581f819b4d177b4/src/lmic/lmic_config_preconditions.h#L52-L60

For users who are not familiar, they can use a BSP (like MCCIs) that supports setting the bandplan from the UI. It is easy to modify any BSP to do this by modifying `boards.txt`; you can refer to how MCCI did it by looking at boards.txt in our BSPs which are also in this project.

I apologize that `ARDUINO_LMIC_PROJECT_CONFIG_H` is not documented, I will file a ticket to document it. But I'm not able to change the configuration scheme at this time.

terrillmoore, 2019-07-31:
See ticket #356.

closed 2019-07-31 by terrillmoore

lnlp, 2019-08-01:
#### Issue has been closed but is not solved.

A library in its default configuration should behave (succeed or fail) consistently for every user.
If successful operation of the library depends on critical user dependent settings, then these must be set by the user explicitly and no default values must be used because critical and user dependent. 
Default values for such parameters will often cause problems, especially when they are defined elsewhere where they are hidden for the user. This will be error prone and potentially cause problems and frustration.

I applaud and respect the good work MCCI-Catena is doing with its continued development of the original Arduino-LMIC library.
What I like much less is that the library in its default configuration does not behave consistently for all users. The library sets us915 as default band plan. This default value is only applicable for users located within the US but is incorrect for almost all other regions (majority of TTN users). 
Setting the default band plan to us915 would not be an issue if that would not cause problems for TTN users located outside the US, but unfortunately it does. 

Band plan is a critical user specific setting (for which no default should be set) that many new (migrating) users are probably not yet aware of, may have been forgotten over time because hidden in a location different from the project source code, or may be silently overwritten by a library update without the user being (made) aware of it.
Setting us915 as default band plan will cause - even for users located outside the US - the library to compile successfully in default/unaltered configuration.
But when the code is run, for users located outside the US the library will use an incorrect band plan default setting and will fail to communicate with the TTN network but without notifying the user of the problem.
The user will assume that their application will run successfully, but different from expectations it will not because of incorrect default band plan. Depending on the situation, it may take the user quite some (wasted) time to locate the actual cause of the problem. This will happen to many (new) users again and again.

I wonder what the motivation is for setting us915 as default band plan, while this setting is incorrect and can cause (described) issues for the majority of TTN users?
Maybe MCCI-Catena's motivation for this is to make usage of Arduino-LMIC easier for customers of its products _(based on assumption that most MCCI-Catena customers are US-based)_. 
If so that would explain the choice for setting us915 as default band plan. In that case I would like to point that there are more elegant ways to implement this than forcing every user globally with us915 as (hidden) default band plan.

MCCI-Catena Arduino-LMIC informally has become the new defacto Arduino-LMIC library.
Many (even experienced) users switching from the original Arduino-LMIC to MCCI-Catena's enhanced version will assume that they can use it as a drop-in replcement but when doing so they will run into the above described problems caused by us915 set as default band plan. This will result in a less fluent migration and less positive first experience which can be prevented by not setting a default band plan.

**Please do not set a default band plan. 
Instead notify the user that it has to be set explicitly _(when not yet done)_.**

_(A simple but efficient solution for implementing this was suggested in my initial post.)_

An additional advantage of not setting a default band plan is that the user will be notified when the band plan setting was previously changed but is later (silently) overwritten by a library update.

---

#### Feedback on suggested workarounds/alternatives:

Drawbacks of current implementation of library configuration settings (band plan) are caused by:

- The location of lmic_project_config.h in library source code location _(described in earlier post)_.
- The fact that us915 is set as default bandplan selection _(described above)_.

Two alternatives have been suggested as workarounds:

1. _From different issue thread:_ Define ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS (on build command line) to prevent that lmic_project_config.h located in the library source code location is used.

2. _From this thread:_ Explicitly define ARDUINO_LMIC_PROJECT_CONFIG_H=\<my project directory\>/lmic_project_config.h (on build command line) to use a different instance of lmic_project_config.h than the one located in the library source code. _(New or previously undocumented.)_ 

Feedback:

* Both alternatives do not work for Arduino IDE.
While most less experienced/less advanced users will use Arduino IDE.

* Both alternatives work for more advanced IDEs (e.g. PlatformIO) but user will explicitly have to configure the project for this.

* The second alternative (ARDUINO_LMIC_PROJECT_CONFIG_H) assumes a certain way of working. Requires that project related configuration settings (dedicated for LMIC library or combined with other) are maintained in a header file. This is not necessarily the preferred way for maintaining project settings for some of the more advanced IDEs.

* The second alternative (ARDUINO_LMIC_PROJECT_CONFIG_H) does not add substantial improvement to the first (ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS).
The first alternative causes the library not to define a band plan at all and ignore lmic_project_config.h so it can be freely defined in a different way.
[... cut ...]
* None of the suggested alternatives fixes the problems caused by having us915 defined as default band plan, at least not for a default installation of the library without any changes added to the project configuration.

* The issues caused by defining us915 as default band plan have not been solved but the issue has been closed. These issues which impact all users located outside of the US (the majority of TTN users) can be simply prevented by not setting a default band plan and letting the user explicitly set a bandplan instead (as described above).