mcci-catena/arduino-lmic #353: Powering radio down between transmissions

Mark-Wills, 2019-07-22:
Good afternoon,

I have been using the LMIC library on an ATMEGA based design that I am currently prototyping. For ultra-low-power (around 7uA) my design turns of the radio (and other hardware) using a MOSFET so that the radio is physically powered off, not just in a sleep mode and places the CPU in a low-power sleep mode.

I'm wondering how I work that in the with the LMIC library, and indeed with the LoRaWAN protocol in general. Will I have to save some state somewhere, power the radio down, and then restore that state prior to transmitting? I presume that the library itself knows the keys, DEVUI and packet number etc. However, I am worried that things like register settings in the radio IC (SX1276) will be lost between power cycles.

If I call LMIC_reset() upon wake up will that be enough to re-initialise the radio in terms of register configuration? How would I restore things like packet number, or is that retained in the library? What items of state (if any) would I need to restore, and how?

Any guidance and insight would be highly appreciated.

With thanks

Mark

cstratton, 2019-07-22:
Is this really worthwhile when the radio data sheet implies 1 uA consumption in sleep mode when properly configured?  If you are going to do it, remember not to leave a voltage on any I/O running to the radio when you power it down, otherwise you will partially re-power it through the I/O in an out of spec way.

In terms of needed radio state, functions in radio.c such as txlora() and rxlora() seem to be pretty comprehensive, my personal recommendation would be that you start there, and if you find there's any other setting they are not covering, add it - but they may already be comprehensive.

Reseting LMIC seems like something best avoided, at least so long as your MCU uses sleep modes that can retain program state.

You may want to avoid powering down the radio between transmit and the corresponding RX window(s), or at least if you do so, you might need to hack in a few extra milliseconds to activate your FET switch and bring up the power in a fully cold start, as the receive windows are precisely timed after the end of the corresponding transmission.

Mark-Wills, 2019-07-31:
Chris,

Ah, yes, you're quite right. I should have paid more attention to the datasheet. So, my next question is: Does the library put the device into the low power (1uA) sleep state after transmitting and/or receiving? I don't recall seeing an API call to put the unit to sleep, so I assume the library takes care of this?

Regards

Mark

terrillmoore, 2019-07-31:
Hi @Mark-Wills, yes, the library leaves the radio in sleep mode when not using it.

closed 2019-07-31 by terrillmoore