mcci-catena/arduino-lmic #362: class c is ready in next release?

halukmy, 2019-08-14:
thanks for now

terrillmoore, 2019-08-15:
I want to fix #87 as part of the next release. There are other issues for compliance, but this is the major blocking issue. Once #87 is fixed, class C should be easy and straightforward. Thanks.

terrillmoore, 2019-09-04:
Now that Class A support is essentially done, I plan to make a prerelease 3.0.99, then will start on Class C and multicast. They might even make it into 3.1.0 (current target official release version)

shmrymbd, 2020-02-18:
pls support class C

halukmy, 2020-02-19:
@terrillmoore  CLASS C IS READY?

terrillmoore, 2020-02-19:
Not ready. I got the flu at the Things Conference in Amsterdam and was out for two weeks. Way behind on the day job. It will be a few weeks, I think. And #523 (and related) have to get fixed at a very high priority. Sorry!

terrillmoore, 2020-02-19:
(The good news is that #87 is fixed, which was an important pre-requisite.)