mcci-catena/arduino-lmic #366: Typo in arduino_lmic.h

lnlp, 2019-08-15:
Appears like a syntax error to me, but the pre-processor thinks otherwise and does not complain.
But the space between '#' and 'define' at least looks inelegant.

From arduino_lmic.h:
```cpp
#pragma once

#ifndef _ARDUINO_LMIC_H_
# define _ARDUINO_LMIC_H_
```

lnlp, 2019-08-16:
Won't fix for what reason?

cstratton, 2019-08-17:
Probably because this appears to be an _intentional_ formatting to track the level of #ifdef type controls.  It's actually a rather clever idea, I might borrow it...

lnlp, 2019-08-17:
The C++ preprocessor accepts white space between the # character and the processing instruction (as shown below). But while syntactically valid, that does not mean that one should use it.

```cpp
#define symbol               // standard syntax, commonly used, 'by the book'.
#           define symbol    // valid but uncommon, confusing, hard to read at a glance.
```
One should use common standard syntax that is widely used.
One should not use personal conventions in the form of uncommon non-standard  syntax because it makes code harder to read and understand for others.

For visualization of nesting levels standard indenting can be used, also for preprocessor directives:
_(And when used, it should be applied consistently throughout the code.)_
```cpp
#ifndef symbol
    #define symbol
    ...
#endif 
```

closed 2019-08-17 by terrillmoore