mcci-catena/arduino-lmic #377: [issue87-devel] confirmation seems broken

cyberman54, 2019-08-19:
With current `issue87-devel` branch confirmation seems broken. Tested with EU868 and TTNv2, after sending a confirmed packet this is never being acked by the device; causing TTN to repeat the packet again and again.

terrillmoore, 2019-08-19:
Thanks for testing. Yes, indeed, it seems I broke that sometime late Saturday night while chasing a problem with replicated unconfirmed uplinks, but didn't realize that when I committed the code (wanted a backup copy). I'll try to fix this later today.  (Problem is that the tests take several hours to run, and I get impatient; don't want to commit a possibly broken state, and don't want to stop development while waiting for results. Need a more-finely-branched workflow I guess.)

terrillmoore, 2019-08-21:
Hi @cyberman54 -- I think this is now fixed on the `issue87-devel` branch. Interestingly, the EU868 test didn't catch this (although it was wrong in the logs); but the US915 test does catch it.

cyberman54, 2019-08-21:
@terrillmoore Thanks for the ping. I did a quick re-test with TTNv2 and EU868 and can confirm the confirmation is now working. 
Maybe Redwood needs a bugfix for their EU868 test scheme?

closed 2019-08-21 by cyberman54

terrillmoore, 2019-08-21:
@cyberman54 thanks for retesting!  It seems that this is a hole in the certification tests -- some Certified testers check, others don't; and the test spec isn't clear. There are similar holes in USB test specs. In any case, it's hard to find all device bugs in a test procedure; there are many more ways for devices to be broken than to be correct. RWC just gave me a new version of tester firmware to test with, it will be interesting to check it. Maybe later tonight.

commit d7d8f358 (2019-08-21): Fix #377: don't repeat uplink seqnos for acks of confs

The counter `LMIC.txCnt` was not being reset at the right time; thus
we sometimes treated an ACK or a POLL as a retry. With the latest
changes, even more so.