mcci-catena/arduino-lmic #379: Add compliance script to builds in Travis.

terrillmoore, 2019-08-22:
Shouldn't we just build all examples?

FlorianLudwig, 2019-08-22:
The basic mac stack from semtech does contain actual compliance tests by running the stack inside an emulator and testing against specification with a simulated lora transceiver.

The diff between this version of LMiC and Semtech's is quite big but structurally the common root is clearly visible.  Would you be open to pull requests that bring the stacks closer together?

terrillmoore, 2019-08-22:
@FlorianLudwig Thanks for your comment. I think this is off topic for this issue, which is about building all the example scripts, which we're currently not doing.  GitHub issues are a bad forum for bikeshed discussions, I've been meaning to set up a discourse site, but.. no time.

commit 27687275 (2019-08-24): Fix #379: add CI checks for compliance sketch
commit 33886536 (2019-08-24): Fix #379: add CI checks for compliance sketch
closed 2019-08-24 by terrillmoore