mcci-catena/arduino-lmic #431: Error in documentation

lnlp, 2019-09-06:
In the documentation:

> **2.2.15** ostime_t **us2osticks**(s4_t sec)

should read:
> **2.2.15** ostime_t **sec2osticks**(s4_t sec)

Relates to LMIC-v3.0.99 and previously LMIC-v2.3 .docx/.pdf.

_Adding a document revision chapter would be useful._

terrillmoore, 2019-09-10:
Thanks for your careful review.

Issue 1. Agreed.
Issue 2. Docs can be improved, but the code does not require that you supply an `onEvent` function. See my comments under #433.
Issue 3. Agreed
Issue 4. Agreed
Issue 5. Agreed, will try to clarify.
Issue 6. Agreed.
Issue 7. Yes, we should avoid duplication of that kind. 
Issue 8. See my comments under #433. It is *not* required. The only thing additional if you `#define` it is that a few bytes are saved.
Issue 9. Agreed.

Suggestion. Agreed; I have tools to automatically generate a decent Word redline, and I'll do that once the document stabilizes.

terrillmoore, 2020-12-06:
I can't find the duplication of EV_LINK_DEAD.

terrillmoore, 2020-12-06:
Hi @lnlp -- because of the delay in making the release due to #635, there's a chance to review the changes, if you have time...

lnlp, 2020-12-07:
Hi @terrillmoore,
I'll have to skip it at the moment, sorry.

commit 30de5189 (2020-12-14): Update docs (fix #431).
closed 2020-12-14 by terrillmoore