mcci-catena/arduino-lmic #432: EV_JOINED never triggered

Spomky, 2019-09-07:
Hi,

The example using ABP works fine on a Heltec Lora v2 and packets are correctly received on my TTN console.
When using it with OTAA, I can see activation packets that indicate the node is correctly registered, but the `EV_JOINED` is never triggered and thus no payload is sent.

Note: for information, the working PIN map I use for the Heltec Lora v2 is the following one:

```c
const lmic_pinmap lmic_pins = {
  .nss = 18,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = 14,
  .dio = {26, 34, 35},
};
```

cyberman54, 2019-09-07:
Wrong pin mapping.

change
`.dio = {26, 34, 35},`
to
`.dio = {26, 35, 34},`

Spomky, 2019-09-07:
…no effect….
The activation packets are still there, but no payload is received.
The console displays:

```
Starting
Packet queued
2968: EV_JOINING
328911: EV_TXSTART
4691684: EV_TXSTART
9025098: EV_TXSTART
14003346: EV_TXSTART

```

cyberman54, 2019-09-07:
Then check if you have a Heltec V2 board, or if it's a V1 (which has other pin mapping).
This sure won't be an issue of this library, since i am using this code on different Heltec boards, and all work.

Spomky, 2019-09-07:
I am sure it is a v2. It is clearly written on the board. With the v1 pin mapping nothing works.
With the v2 pin mapping it works fine with ABP but not with OTAA. So no doubt for the version of the board.

The gateway is a brand new [TTIG](https://connectedthings.store/gb/lorawan-gateways/the-things-indoor-gateway-868-mhz.html) so 8 channels.

Do you think it may come from the `project_config/lmic_project_config.h` file?

```c
#define CFG_eu868 1
#define CFG_sx1276_radio 1
#define LMIC_USE_INTERRUPTS //Tryed with and without this line
```

cyberman54, 2019-09-07:
Don't use interrupts as long as you have not the current version of this library. The function was broken for ESP32 in the past.

Spomky, 2019-09-07:
Unfortunately, nothing changes with and without the interrupts

cyberman54, 2019-09-07:
Look here:

https://github.com/mcci-catena/arduino-lmic/issues/277

Spomky, 2019-09-07:
That’s great thank you!
I added the line `LMIC_setClockError(10 * MAX_CLOCK_ERROR / 100);` after `LMIC_reset();` and it works fine now.

I am not sure to clearly understand what’s the behaviour of this function but it solves the issue.
Thank you so much.
Would you like I create a PR in the doc to mention that for this board?

closed 2019-09-07 by Spomky

cyberman54, 2019-09-07:
This is a workaround. Your application is likely causing timing problems breaking the lmic timing.

Spomky, 2019-09-07:
Well… I just opened the `ttn-otaa` example, changed the PIN mapping, app eui, dev eui and app key values.
Nothing more, nothing less.

cyberman54, 2019-09-07:
Yes, but using a Heltec board means you're obviously using ESP32 with ESP32-arduino framework, thus running LMIC on top of freeRTOS. The LMIC library is intended to run on single core systems without RTOS. So timing can be different.