mcci-catena/arduino-lmic #457: US join timing is still too fast

terrillmoore, 2019-09-14:
Unlike the EU-like regions, US joins don't seem to be delayed at all.  And it looks like we allow uplink TXs right away after a JoinAccept; we should be waiting the inter-packet delay to allow the gateway receiver time to recover from the previous downlink transmit, which tends to overload the front end of the receiver.

OstertagM, 2019-09-23:
Hi I wondering why i get sometimes an downlink sometimes not.
I see that the delay in German is around 320ms after uplink is received.
So perhaps this is same? I don't know

terrillmoore, 2019-09-23:
@OstertagM sorry you're having problems. Can you give more info (platform, software versions, etc.)?  

Also your comment is confusing; the downlink delay is fixed by the network (down is "network to device" in lorawan), and is always 1 second, 2 seconds or 5 seconds/6 seconds. But if I read "uplink" as "message to device" then this could be related.

I have seen empirically that some gateways take up to 500 ms to recover when switching from downlink (tx) to uplink (receive) -- especially when receiving on the same channel as tx, as is usually the case in Europe.

OstertagM, 2019-09-24:
I really mean downlink (network to device).
I scheudle an downlink test data in my console. When I see in my gateway (own one) that the downlink is sended after around 330ms after receiving the uplink from my device. Now I don't know who long the downlink is open on my node after sending. Also I don't know if my node checked the packet number. 
It seems that is only to receive the first down packet. After I received one and I scheudle more when all will be received. How can I Debug?

terrillmoore, 2019-12-30:
> When I see in my gateway (own one) that the downlink is sended after around 330ms after receiving the uplink from my device. 

I think this means that the gateway got the packet to be downloaded at 1 second. I don't think the gateway tried to send the packet at 330 ms. If the network is really trying to schedule a downlink at TxEnd + 330ms, then the network is messed up. The LMIC will never receive it.