mcci-catena/arduino-lmic #458: strange behaviour with ACK

thl-mot, 2019-09-18:
I am using a slightly midified compliance-sketch. 
Main changes are UART.Interface to set the LoRaWAN registration parameters.
I temporarely published the sketch on https://bitbucket.org/thl-mot/lmic-gpstracker
I am testing with ttn.
What I see:
1. Device joines correctly
2. Than it tries multiple times by increasing SF to send data 
3. after one recognized fail
4. it send data as expected

`------------------------------------------------------------------------------ 
compliance-otaa-halconfig.ino
LMIC version 3.0.99.5 configured for region 1.
Remember to select 'Line Ending: Newline' at the bottom of the monitor window.
------------------------------------------------------------------------------

calibration not supported
Packet queued
317538 (5080 ms): EV_JOINING
534839 (8557 ms): EV_TXSTART: ch=0 rps=0x01 (SF7 BW125 CR 4/5 Crc IH=0), datarate=5, opmode=88C, txend=534946, avail=0
850162 (13602 ms): EV_RXSTART: freq=868.1 rps=0x81 (SF7 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=88C, txend=538732, avail=0, delta ms 4982
855810 (13692 ms): EV_JOINED: ch=0
netid: 19
devaddr: 2601268D
artKey: 80-2D-02-69-14-73-93-8C-36-E7-99-F3-2C-0E-C9-8B
nwkKey: 73-A4-94-2D-DA-91-19-DB-76-C3-E9-6A-90-29-26-BC
855834 (13693 ms): EV_TXSTART: ch=0 rps=0x01 (SF7 BW125 CR 4/5 Crc IH=0), datarate=5, opmode=888, txend=855799, avail=0
921720 (14747 ms): EV_RXSTART: freq=868.1 rps=0x81 (SF7 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=888, txend=860043, avail=0, delta ms 986
984542 (15752 ms): EV_RXSTART: freq=869.5 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=888, txend=860043, avail=0, delta ms 1991
1273307 (20372 ms): EV_TXSTART: ch=1 rps=0x01 (SF7 BW125 CR 4/5 Crc IH=0), datarate=5, opmode=888, txend=1273411, avail=986555
1339192 (21427 ms): EV_RXSTART: freq=868.3 rps=0x81 (SF7 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=888, txend=1277515, avail=986555, delta ms 986
1402014 (22432 ms): EV_RXSTART: freq=869.5 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=888, txend=1277515, avail=986555, delta ms 1991
1690782 (27052 ms): EV_TXSTART: ch=2 rps=0x02 (SF8 BW125 CR 4/5 Crc IH=0), datarate=4, opmode=888, txend=1690887, avail=1505172
1760331 (28165 ms): EV_RXSTART: freq=868.5 rps=0x82 (SF8 BW125 CR 4/5 NoCrc IH=0), datarate=4, opmode=888, txend=1698526, avail=1505172, delta ms 988
1823025 (29168 ms): EV_RXSTART: freq=869.5 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=4, opmode=888, txend=1698526, avail=1505172, delta ms 1991
2461858 (39389 ms): EV_TXSTART: ch=3 rps=0x02 (SF8 BW125 CR 4/5 Crc IH=0), datarate=4, opmode=888, txend=2461963, avail=1964960
2531407 (40502 ms): EV_RXSTART: freq=867.1 rps=0x82 (SF8 BW125 CR 4/5 NoCrc IH=0), datarate=4, opmode=888, txend=2469602, avail=1964960, delta ms 988
2594101 (41505 ms): EV_RXSTART: freq=869.5 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=4, opmode=888, txend=2469602, avail=1964960, delta ms 1991
3232934 (51726 ms): EV_TXSTART: ch=4 rps=0x03 (SF9 BW125 CR 4/5 Crc IH=0), datarate=3, opmode=888, txend=3233039, avail=2647561
3309171 (52946 ms): EV_RXSTART: freq=867.3 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=3, opmode=888, txend=3247110, avail=2647561, delta ms 992
3371609 (53945 ms): EV_RXSTART: freq=869.5 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=3, opmode=888, txend=3247110, avail=2647561, delta ms 1991
4647210 (74355 ms): EV_TXSTART: ch=5 rps=0x03 (SF9 BW125 CR 4/5 Crc IH=0), datarate=3, opmode=888, txend=4647315, avail=3516032
4723446 (75575 ms): EV_RXSTART: freq=867.5 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=3, opmode=888, txend=4661385, avail=3516032, delta ms 992
4785884 (76574 ms): EV_RXSTART: freq=869.5 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=3, opmode=888, txend=4661385, avail=3516032, delta ms 1991
6061486 (96983 ms): EV_TXSTART: ch=6 rps=0x04 (SF10 BW125 CR 4/5 Crc IH=0), datarate=2, opmode=888, txend=6061591, avail=4884699
6149689 (98395 ms): EV_RXSTART: freq=867.7 rps=0x84 (SF10 BW125 CR 4/5 NoCrc IH=0), datarate=2, opmode=888, txend=6087245, avail=4884699, delta ms 999 
6211744 (99387 ms): EV_RXSTART: freq=869.5 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=2, opmode=888, txend=6087245, avail=4884699, delta ms 1991
sendComplete: uplink failed
8634382 (138150 ms): EV_TXSTART: ch=7 rps=0x04 (SF10 BW125 CR 4/5 Crc IH=0), datarate=2, opmode=888, txend=8634467, avail=6370184
8722586 (139561 ms): EV_RXSTART: freq=867.9 rps=0x84 (SF10 BW125 CR 4/5 NoCrc IH=0), datarate=2, opmode=888, txend=8660142, avail=6370184, delta ms 999
8784641 (140554 ms): EV_RXSTART: freq=869.5 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=2, opmode=888, txend=8660142, avail=6370184, delta ms 1991
8787099 (140593 ms): EV_TXCOMPLETE: ch=7 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), txrxFlags=0x60, FcntUp=0001, FcntDn=0000, txend=8660142, avail=8818346
Lat: 4903669 Lon: 844925
Packet queued
11207239 (179315 ms): EV_TXSTART: ch=0 rps=0x04 (SF10 BW125 CR 4/5 Crc IH=0), datarate=2, opmode=888, txend=11207343, avail=8818346
11295442 (180727 ms): EV_RXSTART: freq=868.1 rps=0x84 (SF10 BW125 CR 4/5 NoCrc IH=0), datarate=2, opmode=888, txend=11232998, avail=8818346, delta ms 999
11357497 (181719 ms): EV_RXSTART: freq=869.5 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), datarate=2, opmode=888, txend=11232998, avail=8818346, delta ms 1991
11368511 (181896 ms): EV_TXCOMPLETE: ch=0 rps=0x83 (SF9 BW125 CR 4/5 NoCrc IH=0), txrxFlags=0xA2; Received ack, FcntUp=0002, FcntDn=0001, txend=11232998, avail=11399759
Lat: 4903669 Lon: 844924
Packet queued
13780115 (220481 ms): EV_TXSTART: ch=1 rps=0x01 (SF7 BW125 CR 4/5 Crc IH=0), datarate=5, opmode=888, txend=13780219, avail=11399759
13846000 (221536 ms): EV_RXSTART: freq=868.3 rps=0x81 (SF7 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=888, txend=13784323, avail=11399759, delta ms 986
13849460 (221591 ms): EV_TXCOMPLETE: ch=1 rps=0x81 (SF7 BW125 CR 4/5 NoCrc IH=0), txrxFlags=0xA1; Received ack, FcntUp=0003, FcntDn=0002, txend=13784323, avail=13880710
Lat: 4903667 Lon: 844923
Packet queued
15724486 (251591 ms): EV_TXSTART: ch=2 rps=0x01 (SF7 BW125 CR 4/5 Crc IH=0), datarate=5, opmode=888, txend=14197695, avail=13880710
15790371 (252645 ms): EV_RXSTART: freq=868.5 rps=0x81 (SF7 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=888, txend=15728694, avail=13880710, delta ms 986
15793831 (252701 ms): EV_TXCOMPLETE: ch=2 rps=0x81 (SF7 BW125 CR 4/5 NoCrc IH=0), txrxFlags=0xA1; Received ack, FcntUp=0004, FcntDn=0003, txend=15728694, avail=15825080
`

``
3. ttn receives the data, and seams to send the ACK

thl-mot, 2019-09-18:
By the way, the distance between gateway and device is about 1m

terrillmoore, 2019-09-18:
Sorry you're having problems.

Can you please send us a link to Amazon or whatever for purchasing a device? We are not seeing this kind of problem using Murata modules (non-ESP32), so my guess is that there's something different on timing based on the FreeRTOS integration..

The log shows that the device is behaving correctly, given that no messages are received. A correlated gateway log (showing downlink speeds) would be helpful. Unfortumately we'd need to ask you to also send the serial log.

Are you using LMIC_setClockError()? If so, what setting are you using?

thl-mot, 2019-09-19:
I am using this device: https://primalcortex.wordpress.com/2017/11/24/the-esp32-oled-lora-ttgo-lora32-board-and-connecting-it-to-ttn/

LMIC_setClockError(1 * MAX_CLOCK_ERROR / 1000);

thl-mot, 2019-09-19:
I also assume that the behaviour curresponds with activation of ACK. I did't see that behaviour yesterday where I didn't send the data with ACK.

By the way, I discovered that every second try fails (see attached log below)

The gateway (TTN) shows one uplink, two downlink messages: SF12BW125, the neither downlink arrives in the device log.

[send.log](https://github.com/mcci-catena/arduino-lmic/files/3630988/send.log)

terrillmoore, 2019-09-22:
The downlink messages are very likely to be network messages, not application messages. The TTN Console doesn't show the details of these messages.

I would try setting the clock error to 5% to see whether that helps. I would also try removing the call to `LMIC_setClockError()`, and see if anything changes.

It is very curious that your timestamps are all very negative in the second log. On a normal Arduino platform, that indicates that millis() has overflowed, as it usually starts at zero. Your second log also shows use of 8 channels, which suggests that you received a downlink that setup the extra channels.  The second link doesn't show any sign to me that RX succeeded. It's helpful to give succeeding logs as well as failing logs, to enable diagnosis.

We've not seen behavior like this -- but we've heard complaints about "RF quality" on the ESP32 boards. I'm not sure whether I'll be able to help without getting one of the boards.

terrillmoore, 2019-09-22:
Unfortunately, the referenced webpage doesn't give a link to the seller, and there's lots of evidence that there are lots of similar boards from many vendors. I can't really help without a confirmed link to a reliable source. Thanks.

thl-mot, 2019-09-24:
As each of my boards show different behaviour, I now switched to my own design. Its just a plain esp32 (WROOM-32U) and a RFM96 working on 868MHz. The connections match `Arduino_LMIC::GetPinmap_ttgo_lora32_v1();`.

There is still one thing that seams to be strange:
`Packet queued`
`12775757 (204412 ms): EV_TXSTART: ch=4 rps=0x01 (SF7 BW125 CR 4/5 Crc IH=0), datarate=5, opmode=888, txend=11115182, avail=10931987`
`12841642 (205466 ms): EV_RXSTART: freq=867.3 rps=0x81 (SF7 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=888, txend=12779965, avail=10931987, delta ms 986`
`12906638 (206506 ms): EV_RXSTART: freq=869.5 rps=0x86 (SF12 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=888, txend=12779965, avail=10931987, delta ms 2026`
`13193233 (211091 ms): EV_TXSTART: ch=5 rps=0x01 (SF7 BW125 CR 4/5 Crc IH=0), datarate=5, opmode=888, txend=13193340, avail=13084410`
`13259117 (212145 ms): EV_RXSTART: freq=867.5 rps=0x81 (SF7 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=888, txend=13197440, avail=13084410, delta ms 986`
`13324113 (213185 ms): EV_RXSTART: freq=869.5 rps=0x86 (SF12 BW125 CR 4/5 NoCrc IH=0), datarate=5, opmode=888, txend=13197440, avail=13084410, delta ms 2026`
`13610709 (217771 ms): EV_TXSTART: ch=6 rps=0x02 (SF8 BW125 CR 4/5 Crc IH=0), datarate=4, opmode=888, txend=13610816, avail=13375359`
`13680257 (218884 ms): EV_RXSTART: freq=867.7 rps=0x82 (SF8 BW125 CR 4/5 NoCrc IH=0), datarate=4, opmode=888, txend=13618452, avail=13375359, delta ms 988`
`13745125 (219922 ms): EV_RXSTART: freq=869.5 rps=0x86 (SF12 BW125 CR 4/5 NoCrc IH=0), datarate=4, opmode=888, txend=13618452, avail=13375359, delta ms 2026`
`Received message on port 0: 0 bytes`

If I'm Interpreting this log correctly. It seams that my device sends the data 3 times. On my (professional) LoRa platform (Zenner IoT Element IoT) there is only one up-transmission. 
This does not happen too often. But I can see it quite some times when debugging my code.
If the esp32 device just had timing problems receiving the downlink-messages I would expect 3 UP-Messages and three Down messages, as my device triggers 3 valid ups. 
So my guess is, that either the transmission from my device is corrupted (bad antenna, ...) or the transmission is disturbed by some other lora or radio device.

terrillmoore, 2019-10-10:
It looks like a confirmed message that is not getting acknowledged. (The data rate pattern is diagnostic of retries for a confirmed uplink.)