mcci-catena/arduino-lmic #474: Setting RX2 download for TTN

terrillmoore, 2019-10-10:
When using The Things Network in the EU868 region, you must set the RX2 Data Rate to SF9 (DR4).  It turns out not to be as simple as it should be. We need a better way of tailoring LMIC for pre-JOIN (and ABP) network parameters.
 
@cyberman54 wrote:

> Moved a small step forward here. Found out why in my code the setting RX2 == SF9 for ttn did not work. I called `LMIC.dn2Dr = EU868_DR_SF9` right before `LMIC_startJoining()` but this can't work*. After i moved `LMIC.dn2Dr = EU868_DR_SF9` to the event handler (put it to case JOINING, as found in the compliance script) i now see correct setting SF9 for RX2 in the log while joining. Perhaps this should be documented somewhere, otherwise probably most of TTN & LMIC users will use wrong RX2 setting.
>
> *) Because `LMIC_startJoining` is calling `resetJoinParams()` which overwrites
> LMIC.dn2Dr       = DR_DNW2;

_Originally posted by @cyberman54 in https://github.com/mcci-catena/arduino-lmic/issues/455#issuecomment-540459419_

jpmeijers, 2019-10-15:
It should not be necessary to set the `dn2dr` when doing OTAA, as the network will configure it during join.

For ABP using `LMIC.dn2Dr = EU868_DR_SF9` should be fine as there is no join process to override this.

manuelbl, 2019-10-22:
I've come to the same conclusion. Also see this [issue](https://github.com/TheThingsNetwork/ttn/issues/155). It discusses the same question. And the conclusion is a commit that changes the data rate to SF12 (for join).

terrillmoore, 2019-10-22:
> a commit that changes the data rate to SF12 (for join).

I think we're all violently agreeing. It should not be done after join; but pre-join, you're not going to have a good TTN experience without setting LMIC.dn2dr. The point of this issue is that there needs to be an API to take care of this (and other network-specific pre-join parameters).

manuelbl, 2019-11-08:
Do you have any hard evidence that SF9 is used for _JoinAccept_ in Europe?

I just stumbled across this [statement](https://www.thethingsnetwork.org/forum/t/node-works-but-does-not-switch-to-sf7/31543/5) from Jac Kersing (re EU868):

> And TTN does send OTAA replies using the standard RX2 parameters as documented in the LoRaWAN standards document

Standard RX2 parameter is SF12.

jpmeijers, 2019-11-08:
(Note that my comments are only relevant for EU868.)

Yes, have a read through https://github.com/TheThingsNetwork/ttn/issues/767 and also note that I made a change to TTN's v2 code base - a change that is running in production since 15 October 2019.

For OTAA:
* Join accept in RX2 will be sent using standard SF12
* During the join the device is configured for TTN RX2 with SF9
* From then on forward TTNv2 will always use SF9 for RX2

For ABP:
* TTNv2 tries using SF9 for RX2
* If the ADR acks are not received by TTNv2 it will fall back to using SF12 for RX2 and stay there - never trying SF9 again

terrillmoore, 2019-11-08:
Closing this as being mistaken.

closed 2019-11-08 by terrillmoore