mcci-catena/arduino-lmic #484: Define a channel and SF to join

Beliwars, 2019-10-22:
Hello guys, 

I'm trying to define a fix channel frequency and SF when starting a TTN join; I see my device it's only joining when using freq 868.1 MHz and SF=8 but the join takes a while because the library makes a round of all channels on SF=7 before trying SF8; i tried to use this:

LMIC_setDrTxpow(EU868_DR_SF8,14);

before the do_send job, but it appears to do nothing, round robin starts from SF=7, I'm on a EU868 region.

Do you know is there is any way I can speed up the join process by defining which channel and spreading factor to use?

Thanks in advance for your support

jezdd, 2019-10-22:
I had mine here >>

    case EV_JOINING:
        LMIC_setDrTxpow(DR_SF7, 14);
        LMIC_setLinkCheckMode(ADR_MODE);
        LMIC_setAdrMode(ADR_MODE);
        Serial.println(F("EV_JOINING"));
        break;

Beliwars, 2019-11-13:
Hello @jezdd thanks for you reply, in the end we changed the GW config the device joins with SF=7 without any issues so I didn't try your suggestion.

regards.

closed 2019-11-13 by Beliwars