mcci-catena/arduino-lmic #486: Reset in case of failure

geppoleppo, 2019-10-22:
Hi,
one question; there is a simple (more or less) method for reset MCU in case of failure?
Thanks

cstratton, 2019-11-27:
A software-triggered reset is a platform dependent functionality that is not really portable.  Typically LMiC fails in an assert() which in most cases will be an infinite loop.  If your platform has a configured watchdog that could result in a reboot, if not you may want to come up with your own way of forcing a restart.

Crash-restarts of LoRaWAN nodes are substantially problematic, as both the frame counter and any past used join nonces have rollback protection server-side, and if either uplink or downlink frame count has rolled over the 16-bit boundary multiple times, an at least approximate starting value of the full 32-bit counter must be known or checksums will fail.

Ideally paths that result in an assert() from things like unexpected radio state (due to an SPI glitch or whatever) would instead trigger a warm restart - fully reconfiguring the radio from scratch.   Keeping copies of the key registers somewhere like the RTC backup domain (or at the tail end of RA with the startup code modified to not erase that) and then sanity checking on startup could be good.  If a given platform is using a non-volatile memory well to track these things across power loss, then restarts should be accommodated, too.

terrillmoore, 2019-12-30:
> reset MCU in case of failure?

On ARMs, you can call NVIC_SystemReset() to cause a reset. Please bear in mind the conditions mentioned by @cstratton.

terrillmoore, 2019-12-30:
Closing because I think this is answered. Feel free to reopen.

closed 2019-12-30 by terrillmoore