mcci-catena/arduino-lmic #491: How to configure abp/otaa for IN865 band

asprakash, 2019-11-19:
I am using IN865 band in India and setup Node/gateway. I am not clear about the configurations we need to modify for IN865 band on the existing otaa/abp sample codes. Any kind of inputs on this IN 865 band configuration is appreciated.

TIA.

terrillmoore, 2019-11-19:
Vel and Dhinesh are using this package in India in our Chennai office. We've tested with The Things Network and Actility (as well as Tata, I believe). We'll need to know:

- your device type
- your gateway type
- your target network

If you're testing with TTN using an MCCI BSP, it's really easy -- just select IN865 in the Arduino UI. It gets less and less easy as you get further from that, so that's a good place to start if you can; but you basically define `CFG_in865` and everything else will adapt.

asprakash, 2019-11-20:
Hi. Thanks for your input. 

- Device : I am using Arduino Uno & Adafruit feather M0 for POC
- Gateway : I am using RAK831 Shield on RPI
- Network : Chirpstack/loraserver.io
Due to the TTN's FUP and our requirement's constraint, TTN is not suitable for my requirement. I gone through discussion list and other threads on the TTN. There are many suggestions & guidance. But there is no example code for IN865 band as like ttn-abp_in865. If you provide such samples (even it is for MCCI/TTN), its easy for someone to understand the exact porting when we use others networks & hardwares.
TIA.

dhineshkumarmcci, 2019-11-20:
Hi Prakash,
We suggest you to use our example sketch [ttn-otaa.ino](https://github.com/mcci-catena/arduino-lmic/blob/master/examples/ttn-otaa/ttn-otaa.ino) for your testing. You can configure to IN866 by enabling  `#define CFG_in866 1` and on commenting `#define CFG_us915 1` in [lmic_project_config.h](https://github.com/mcci-catena/arduino-lmic/blob/master/project_config/lmic_project_config.h). And this configuration should be helpful.
Let us know how it helps.

asprakash, 2019-11-20:
Hi Dinesh,
Thanks for your input. I will try it out and get back to you with my
feedback.

TIA.

terrillmoore, 2019-11-20:
Hi @asprakash -- btw, I strongly recommend against the use of ABP for most projects. Without going into the details, LoRaWAN is really engineered (in my opinion) for OTAA; it works much better. I think you're on the right path to try the OTAA sketch.

closed 2019-11-20 by asprakash

asprakash, 2019-11-20:
I tried OTAA as per your instructions. I received the log on Arduino. Its
trying to join with the server and I can able see the join request on the
Chirpstack lorawan server Not sure whether there is a problem on ardunio
sketch or server side. Also attached my program. PFA.

Arduino log :
Starting
Packet queued
195: EV_JOINING
118934: EV_TXSTART
501544: Unknown event: 20
769518: EV_TXSTART
1152194: Unknown event: 20
1611562: EV_TXSTART
...
TIA.

dhineshkumarmcci, 2019-11-21:
Hi Prakash,
We like to know the BSP used at your end. So that we can try to reproduce the issue and assist you better.

We like to let you know, Github comments do not support any attachments. Please let us know about the changes made by you in the sketch. 

We also suggest you to check the wiring of Adafruit Feather M0 board. Please refer below sections of Adafruit Feather M0 page:

- [Arduino Wiring](https://learn.adafruit.com/the-things-network-for-feather/arduino-wiring)
- [Arduino Setup](https://learn.adafruit.com/the-things-network-for-feather/arduino-setup)
- [Arduino Code](https://learn.adafruit.com/the-things-network-for-feather/arduino-code)

asprakash, 2019-11-21:
Hi Dinesh,
Thanks for your inputs.

- Right now, I am using both Arduino Uno & Feather M0 along with mcci lmic
library. Using Adafruit BSP from the Arduino board manager.
- Today I tested ttn-abp with IN865 band from Arduino Uno. Modified the
lmic_project_config.h file as per your instruction, Comment #define
CFG_us915 1 and enabled #define CFG_in866 1. No other changes. Also managed
to configure the lmic pin maps as per the board specification.
- Configured NWKSKEY, APPSKEY & DEVADDR.
- I was able to receive the datas on the loraserver. It confirm the mcci
lmic working for IN865 band. However, I am not yet successful with OTAA
method. I am also investigating on the server side whether my configuration
is correct or not. I will give you my feedback once I crosschecked.

TIA.