mcci-catena/arduino-lmic #495: Network and application session keys are printed incorrect in ttn-otaa.ino example

lnlp, 2019-11-25:
In example ttn-otaa.ino on EV_JOINED the application and network session keys are printed but incorrect. The same may hold for other examples (but not checked).

When using the HEX specifier print(value, HEX) will never print a leading zero while it actually should (from perspective of the value's type). So byte value 0x0F will be printed as "F" instead of "0F". As a result below code extract from the example will print the keys incorrectly if any of the individual bytes have a value smaller than 16 (relevant leading zeros are missing for each byte).

```
              Serial.print("artKey: ");
              for (int i=0; i<sizeof(artKey); ++i) {
                Serial.print(artKey[i], HEX);
              }
              Serial.println("");
              Serial.print("nwkKey: ");
              for (int i=0; i<sizeof(nwkKey); ++i) {
                Serial.print(nwkKey[i], HEX);
              }
```

cstratton, 2019-11-27:
Arduino Libraries do sometimes leave much to be desired.  Some approaches are discussed at https://forum.arduino.cc/index.php?topic=38107.0

Actual blame goes here:

https://github.com/arduino/ArduinoCore-API/issues/40

lnlp, 2019-11-27:
Sure but that still makes the example print incorrect keys.

> will never print a leading zero while it actually should
Already put the blame at Arduino's print(). 🙂
But if they (ever) fix this after many years that will probably impact much existing application code.

I did not post alternatives/workarounds as Terry probably will implement his own preferred fix.

terrillmoore, 2019-11-27:
Yeah, sorry for slow response. @lnlp I agree that this is confusing for people, and it's an easy fix. There are a zillion ways to do it but the easy way is to just write a new function. Not very object-oriented, but in these sketches it will probably be just fine.

```c++
static void printHexByte(uint8_t c) {
  if (c < 16)
    Serial.print('0');
  Serial.print(c);
}
```

I believe I'm already doing something like that in the compliance sketch. I'll try to do tis in the next day or two.

terrillmoore, 2019-11-28:
@lnlp please review #497.

closed 2019-11-28 by terrillmoore

lnlp, 2019-11-28:
Only 19 hours between your review request and closing the issue...

terrillmoore, 2019-11-29:
Feel free to reopen. I needed to merge the request to support other work; we can update as needed.

commit 19057c46 (2019-11-28): Fix #495: improve display of AppSKey and NwkSKey