mcci-catena/arduino-lmic #499: AU915 configuration

camilo-17, 2019-12-11:
Hello everyone, excuse my English, I am trying to send Uplink packets with the au915 band it works but packets are lost because I have not been able to configure exactly the frequencies at which my gateway works 

the frequencies  that my gateway works are:

915.2MHz, 915.4MHz, 915.6MHz, 915.8MHz, 916,0MHz, 916.2MHz, 916.4MHz, 916.6MHz, 920.0MHz, 920.2MHz, 920.4MHz, 920.6MHz, 920.8MHz, 921.0MHz, 921.2MHz, 921.4MHz

I have tried to modify the lorabase_au921 file in the downloaded library that I have but I have not been successful, I think the solution goes that way however I do not know very well how to configure it correctly, I would like you to tell me what change I have to do exactly thank you very much for your help.

cstratton, 2019-12-11:
Which of your frequencies are uplink and which downlink?

Your first set seems to match the official spec for the first sub-band of AU921 uplinks

But the second set doesn't match either the spacing or the assignment of downlinks.

Are these all uplinks in the sense of a 16 channel gateway handling two different sub-bands?

Typically in an OTAA case the node blindly tries all sub-bands (though you can pre-emptively disable unsupported channels) and then in the join process and/or ADR responses gets a listing of the actual supported channels from the network server.

But this is all done in terms of enabled/disable channel numbers, not by modifying the raw frequency assignments.

camilo-17, 2019-12-12:
Hello cstratton thanks for your answer. I use ABP activation and it works sends data to my server and I can view the packets however some packets lose and when I modify the frequencys it not works and also I can´t view packets of downlink I tried to modify that frequencies:

```
enum {
        AU921_125kHz_UPFBASE = 915200000,
        AU921_125kHz_UPFSTEP = 200000,
        AU921_500kHz_UPFBASE = 915900000,
        AU921_500kHz_UPFSTEP = 1600000,
        AU921_500kHz_DNFBASE = 923300000,
        AU921_500kHz_DNFSTEP = 600000
};
enum {
        AU921_FREQ_MIN = 915000000,
        AU921_FREQ_MAX = 928000000
};

```

If I modify these frequencies, I could solve my problem? or What procedure I can do for solved that, again excuse me english I tried to support me in traductors and maybe are not 
trustworthy.

terrillmoore, 2020-02-09:
@camilo-17 is this still an active issue?