mcci-catena/arduino-lmic #500: From v 2.3.2 to 3.0.99 issue

mutton74, 2019-12-17:
Hi,

I am an amateur users of your library but have been storing and retrieving LMIC state parameters in RTC memory while in deep sleep (ESP32 based board)

This is to avoid an OTAA join every wake up from deep sleep, and enabling use of ABP.

With an update from 2.3.2 to 3.0.99 I now get errors calling two of these LMIC states (LMIC.ladrAns and LMIC.devsAns)

'struct lmic_t' has no member named 'ladrAns' 
'struct lmic_t' has no member named 'devsAns'

Any idea why that has changed between the versions and if they are required for a successful ABP join after and OTAA join?

thanks in advance
Paul

terrillmoore, 2019-12-21:
Thanks for your note.

They have changed because the LMIC MAC parser was architecturally incorrect, and had to be rewritten. If you check the issues, there's a pretty detailed log. The old architecture didn't maintain the order of responses, and that's needed for LoRaWAN compliance.

They were not long term state; no need to save or restore them. Avoid using any of the LMIC contents that are not explicitly documented as public, as they will change in the future. Although it's possible to do what you're doing, I regret that we can't really guarantee backward compatibility across major versions. The intent is to introduce APIs that you can call to get binary blobs that are opaque to you, and versioned; later you restore them with a different call.

Remember that Semantic Version major change (2 -> 3) means "breaking change".

Best regards,
--Terry

mutton74, 2019-12-28:
Thanks for the info
> They were not long term state; no need to save or restore them
This is all I need to know, will remove them from the saved state in my code

thanks again

Paul

closed 2019-12-30 by terrillmoore

mutton74, 2020-01-04:
Hi Terry,

By not using these two lines of state code, we get an ADR downlink with every uplink forever, not just the usual one or two after joining the first time - any ideas?

terrillmoore, 2020-01-05:
Yes, but in order to comment accurately I need to know:

1. which version of LMIC are you using? Check ARDUINO_LMIC_VERSION
2. which network are you using? (The Things Network, ChirpServer, Actility, etc.)
3. which region are you using?

Thanks,
--Terry

mutton74, 2020-01-05:
Thanks Terry, here are some details

1) Not quite sure how to do that (amateur sorry)

Here is what is in the lmic.h file
`// LMIC version -- this is ths IBM LMIC version
#define LMIC_VERSION_MAJOR 1
#define LMIC_VERSION_MINOR 6
#define LMIC_VERSION_BUILD 1468577746

// Arduino LMIC version
#define ARDUINO_LMIC_VERSION_CALC(major, minor, patch, local)	\
	(((major) << 24ul) | ((minor) << 16ul) | ((patch) << 8ul) | ((local) << 0ul))

#define	ARDUINO_LMIC_VERSION	ARDUINO_LMIC_VERSION_CALC(3, 0, 99, 3)	/* v3.0.99.3 */`

2) TTN
3) AU921 in the project_config file (AU915)

terrillmoore, 2020-01-05:
Yes, that's a known problem in v3.0.99.3.  Please update your library to head, which is v3.0.99.10.  We made another breaking change, which is CFG_au915 is now (correctly) the magic word for Australia. I tried to make backward compatibility work, but... you may run into an issue.

mutton74, 2020-01-18:
> Please update your library to head, which is v3.0.99.10

Sorry Terry, just getting back to this - how do I do the above?

Latest library available within Arduino IDE is v3.0.99.3

thanks in advance

terrillmoore, 2020-01-18:
Ah, sorry. I will release 3.1 later today. You could use github to update; but based on feedback, it’s time to release 3.1.

 

Best regards,

--Terry

terrillmoore, 2020-01-19:
Hi @mutton74  -- because of bug #523, #521, #524 I think it's not a good idea to release.

What you should do is follow the procedure for manually installing arduino libraries -- get a zip file using the GitHub web page, and then unzip that over your existing arduino-lmic library. If you installed from the IDE, it will be `..../libraries/MCCI-LoRaWAN-LMIC-library`. Replace that library with the file you download. Don't leave the old one lying around, it will confuse the Arduino IDE. Alternately, you may need to wait a day or two until I have a change to correct #524. I'm traveling now, so things are even slower than normal.