mcci-catena/arduino-lmic #502: Add precision transmit start option

terrillmoore, 2019-12-24:
For chasing down the SF12 problem, I built a fairly elaborate test based on transmitting with one radio, signaling with a GPIO, and using that to trigger a receive one second later.

It turns out that there's too much jitter in the TX path. Normally not a problem, but for this case, it's an issue. This is useful enough to test that I'm adding a new `os_radio()` opcode: `RADIO_TX_AT`. This causes the driver to use the *input* value of `LMIC.txend`, and strive to launch the transmit at the indicated time. (Otherwise `LMIC.txend` is ignored on entry, and updated on output.) This makes it possible to get precision transmit starts (probably within +/- a few ostime_t clicks, or about 4us, although I don't have accurate enough equipment, or clocks on my devices, to be able to nail it down). I used the scope to measure and adjust so I was getting 1 second transmits, and this let me start examining the SF12 problem.

terrillmoore, 2019-12-24:
See script at https://github.com/mcci-catena/rwc_nst_test to see how this is used.

terrillmoore, 2019-12-26:
This really worked well for me in getting a precision transmit time. Jitter is no longer visible on scope (means it's much less than a millisecond).

cyberman54, 2019-12-26:
This approach could be useful for precise timeofday sync, too.

terrillmoore, 2019-12-26:
> This approach could be useful for precise timeofday sync, too.

Could be. And it could also be very useful for reliably scattering uplinks in time (provide a randomizer to the launch time.)  Anyway, it's like three lines of code, and it's useful for testing. So I think it will be a permanent part of the library. Not sure how it will get accessed from the upper API yet. We'd have to work out how it interacts with retries, etc.

commit f213c432 (2019-12-29): Fix #502: add os_radio(RADIO_TX_AT), version is 3.0.99.8
closed 2019-12-30 by terrillmoore