mcci-catena/arduino-lmic #519: Timing Analysis

manuelbl, 2020-01-05:
I finally had time to complete the [SX127x Probe](https://github.com/manuelbl/sx127x-probe), run sufficient tests, compare LMIC and Semtech code and come up with some findings.

# Probe

The SX127x probe is a STM32 MCU that I can connect to a SX127x chip/board in parallel to the MCU that controls it. It observes the SPI communication (MCU to SX127x only) and the DIO0 and DIO1 pins. Based on the observations, it can measure and analyze many timing parameters – in particular the RX windows that are so essential for joining and downlink packets. The output (on a UART TX pin) looks something like this:

```
  -97691: TX start
       0: TX done
          SF7, 125000 Hz, payload = 48 bytes, airtime = 97536us, ramp-up = 155us
  980791: RX1 start
 1026155: RX1 timeout
          SF7, 125000 Hz, airtime = 45056us, ramp-up = 308us
          Margin: start = 21973us, end = 21035us
          Correction for optimum RX window: -469us
 1986643: RX2 start
 2044296: RX2 timeout
          SF9, 125000 Hz, airtime = 57344us, ramp-up = 309us
          Margin: start = 25336us, end = 23816us
          Correction for optimum RX window: -760us
```

The interesting parameters are the computed *margin* at the start and at the end. It's a measure by how many µs the MCU (or the the gateway) could be early or late and the transmitted packet would still be successfully received. Except for power consumption, the bigger the margin, the better. For the best robustness, the margins at the start and end should be equally long. The conceptual model is described in more detail in the [README](https://github.com/manuelbl/sx127x-probe).

# Findings

## Timing accuracy

The first finding was that the LMIC code is very repeatable in executing an action at a certain time, within about 2µs. The SemTech code (https://github.com/Lora-net/LoRaMac-node) is only accurate within 1ms, about 500 times worse. This is not surprising as all time values are rounded to 1ms internally. As it turns out, it's still sufficiently precise.

## RX window offset and length

LMIC and SemTech use two different formulas to determine when the RX window should start and how long it should be. I've simplified the formulas somewhat (no rounding, no drift). At the core they are:

| Library | Window length [ms] | Window offset [ms] |
| ---- | ---- | ---- |
| LMIC | 6 * *tSymbol* + 2 | -2 |
| SemTech | 4 * *tSymbol* + 40 | 2 * *tSymbol* - 20 - *tWakeUp* |

*tSymbol* is the length of a symbol. It depends on the data rate.

*tWakeUp* is the time the SX127x needs to wake up (process the command to start receiving, stabilize the frequency etc.). It is set to 1ms. According to my observations, the wake up time is about 300µs.

The SemTech window is obviously much longer. Independent of the data rate, they have a total margin of about 40ms while LMIC has only 2ms.

The purpose of the SemTech formula for the window offset isn't obvious at first sight. It has been chosen such that the margins at the start and at the end are equally long. So it seems that SemTech's timing is based on the same conceptual model.

The effective values for the different data rates (for EU868) are shown below. They use the more complex formulas with rounding. And they match what I have observed by hooking up the probe to a circuit running with the respective software.

### SemTech

| DR | SF | BW | tSymbol [ms] | Window length [syms] | Window length [ms] | Offset [ms] | Margin start [ms] | Margin end [ms] |
| ----: |----: |----: |----: |----: |----: |----: |----: |----: |
0 | 12 | 125000 | 32.768 | 6 | 196.61 | 32.00 | 33.24 | 32.00
1 | 11 | 125000 | 16.384 | 7 | 114.69 | 8.00 | 24.47 | 24.38
2 | 10 | 125000 | 8.192 | 9 | 73.73 | -5.00 | 21.08 | 19.58
3 | 9 | 125000 | 4.096 | 14 | 57.34 | -13.00 | 20.89 | 19.77
4 | 8 | 125000 | 2.048 | 24 | 49.15 | -17.00 | 20.80 | 19.86
5 | 7 | 125000 | 1.024 | 44 | 45.06 | -19.00 | 20.75 | 19.91
6 | 7 | 250000 | 0.512 | 83 | 42.50 | -20.00 | 20.72 | 19.42

The slight difference between the length of the start and end margin is due to rounding and due the a more precise model regarding wake up. But given the 1ms resolution of SemTech's code and the 20ms margin, it's not relevant.

### LMIC

| DR | SF | BW | tSymbol [ms] | Window length [syms] | Window length [ms] | Offset [ms] | Margin start [ms] | Margin end [ms] |
| ----: |----: |----: |----: |----: |----: |----: |----: |----: |
0 | 12 | 125000 | 32.768 | 7 | 229.38 | -2.00 | 67.24 | 30.77
1 | 11 | 125000 | 16.384 | 7 | 114.69 | -2.00 | 34.47 | 14.38
2 | 10 | 125000 | 8.192 | 7 | 57.34 | -2.00 | 18.08 | 6.19
3 | 9 | 125000 | 4.096 | 7 | 28.67 | -2.00 | 9.89 | 2.10
4 | 8 | 125000 | 2.048 | 7 | 14.34 | -2.00 | 5.80 | 0.05
5 | 7 | 125000 | 1.024 | 8 | 8.19 | -2.00 | 3.75 | 0.05
6 | 7 | 250000 | 0.512 | 10 | 5.12 | -2.00 | 2.72 | 0.05

The LMIC margins are asymmetric and differ considerably depending on the data rate. For higher data rates, the margin at the end is low.

# Conclusions

Even though the LMIC margins are  narrow, it still worked. But it could make sense to adopt a formula similar to SemTech's, i.e. a more or less fixed and symmetric margin independent of the data rate. It would probably increase robustness without sacrificing low power consumption. 

@terrillmoore I'm interested in your feedback. Is the analysis understandable? Do you see an obvious errors?

terrillmoore, 2020-01-06:
@manuelbl what a great test tool! Thanks for the careful work!

I assume the LoRa device you are testing is using an ESP32 or some other device with a 100 ppm or better system clock, rather than the Murata module (which has a 4000ppm system clock).

No errors that I can see; I came to the same conclusions about timing by analysis, and by my somewhat different approach using [rwc_nst_test](https://github.com/mcci-catena/rwc_nst_test). I can also tell that you are testing with HEAD (3.0.99.10 or later), based on the timing.

You have experimentally confirmed exactly the timing that I designed after finding the results summarized at https://github.com/mcci-catena/arduino-lmic/issues/483#issuecomment-568632897. The key finding was that, with noisy environments, the packet loss rate was much higher if I delayed the start-of-RX by several symbols into the transmit preamble, for example:

https://user-images.githubusercontent.com/19333103/71427763-f7eb7600-2689-11ea-8cab-390b06a7601f.png

This doesn't reproduce, by the way, with pristine environments, but was pretty reproducible in an unintentionally noisy environment.

The comments in the mbed code referenced in #438 (https://github.com/ARMmbed/mbed-os/pull/8822/files) suggest that Semtech was having problems with similar things; they effectively moved their low-speed window into one of the nulls of the picture above.

Based on this, and the observed data integrity problems, I moved the window earlier than Semtech. 

Keeping the window short (in terms of RX syms) is generally good if you can get away with it; the reason Semtech makes it longer is because of the inaccuracy in the millisecond timer you mention. (You have to open up the window if you don't know when it's happening). It's a power thing.

However, we have to also take into account clock accuracy. 100 ppm clocks are typical, but 4000ppm clocks are not uncommon (stm32l0 HSI clocks, for example). This complicates join accept timing, which may be delayed six seconds. 

I did calculations with a 100 ppm clock, and I found that my nominal calculations worked out (I hope someone will double check) but wiht a 4000 ppm clock, the timing blows out (even on the lmic) exactly as described. Please try using `LMIC_setClockError()` to set a 4000 ppm error, and you'll see that the rxwindows at high speed blow up exactly as you show.

If you open the window too late, you won't get enough preamble. I experimentally verified that, using an SX1276 to transmit, the window closes hard 4 symbols after the start of transmit. But the SX1276 is documented as transmitting 10 symbols of preamble by default; and other places state that it requires 6 symbols of preamble to sync up. (So this matches.)  Here's a picture:

[image]

The documentation available is confusing and contradictory, and I couldn't get a good timing reference from the RWC5020, but I believe that the gateways transmit either 8 or 10 symbols of preamble. (I assume that for downlink they'd reduce this if they could since downlink time is relatively precious.)  So my formulas assume that the window is only 8 symbols, and that means we need to land somewhere in the first two symbols.  In fact, because of my experiments, I determined that the radios I have performed the best if the RX windows was already open when the preamble appeared, possibly because the PLLs have stabilized. That biased me to the beginning of the window. I also subtracted 2 ms because it takes 2 ms for the power amps and analog circuits to turn on (again, this isn't well documented, but I found lots of anecdotal evidence; and in the picture above, you can see that integrity starts to go down as you get close to the right hand side of the window (as a starting time). 

Another relevant fact: it actually doesn't matter much if you're a little early, because the hard stop is the end of the preamble. If the SX1276 starts seeing a preamble during the RXSYMS timeout period, it will say "I have a preamble" and will go on to receive the packet.  I didn't experimentally measure how many symbols of overlap you have to have, but I believe it is 4 symbols.

If you aim to land at time t, and the absolute value of your clock error is e (0 being perfect), then you might land at `t_late = (1 + e)*t`.  To avoid this, you need to choose `t_adj = t/(1 + e)`.  (The LMIC approximates this.)  But, in fact, your clock might be fast.  In this case `t_early = (1 - e) * t_adj`, which is `(1-e)/(1+e) * t`.  The LMIC approximates this as `(1-e)*t`, which is optimistic but for e much less than 1, not very optimistic.

So the LMIC, on the assumption that it might be slow moves up `t` to `t_adj`.

Then, on the assumption that it might be fast, it extends the rx window (in units of rxsyms) until it is sure that there is an overlap.

The Semtech and mbed formulas try to "reduce power" by opening the rx window as late as they can (so as to waste less power during preamble time). The fact is, however, that RX downlinks for Class A devices are rare; so it doesn't matter how late you open the window for a successful rx, it matters how *long* you open the windows for failed rx.

terrillmoore, 2020-01-06:
By the way, 

> For the best robustness, the margins at the start and end should be equally long.

I don't agree with this completely, based on my experiments and based on my analysis above. The margins have to be chosen so that worstcase(T early) plus (number of rx syms) overlap the preamble enough to trigger a reception. This is not necessarily symmetrical when you consider the empirical fact that the more preamble you see, the better; and when you consider that there's a fixed (roughly 2ms) time after you start the radio before it actually is receiving reliably.

manuelbl, 2020-01-06:
> For the best robustness, the margins at the start and end should be equally long.

I've tried to come up with a model that takes radio wake up time etc. into account such that 0 margin at the start and at the end is the extreme edge case that still works. However, you have correctly pointed out two things that are not correctly covered in my model:

1. If preamble symbols are detected, the RX windows does not time-out.

2. For the radio wake-up time, I've taken 300µs as it is the difference between the measured timeout (RXSINGLE command to DIO1 signal) and the configured timeout. But I probably have to assume that the timeout timer in the SX127x starts ticking before the radio has reached maximum sensitivity.

I will refine my model accordingly. 

Regarding item 2: You are saying that the radio startup time is roughly 2ms. Do you have any evidence that the timeout starts ticking down before the receiver is really able to detect preamble symbols are that it is only able to detect full symbols? Or how did you come up with 2ms?

Other data is contradictory: I've measured 310µs. SemTech uses 1ms knowing that they will be off by 0.5ms on average. The datasheet shows a figure with a receiver startup time (from standby) and specifies it as TS_FS (60µs) + TS_RE (71µs @ 125kHz) = 131µs only...

(BTW: For the newer SX126x chip, SemTech uses a much longer wake-up time of 8ms! )

And I was wondering why you go to such great length to have precise timing and compensate for clock drift when it could easily be mitigated with a slightly longer RX window. Is it to reduce power compensation?

terrillmoore, 2020-01-07:
> wondering why [do] you go to such great length to have precise timing and compensate for clock drift when it could easily be mitigated with a slightly longer RX window. Is it to reduce power compensation?

Well, to some degree I'm following the existing practice; believe it or not, I'm trying *not* to impose my own design philosophy on the IBM-originated framework. Indeed, the theory is that you want the RX window as short as possible, because you pay the full cost of the RX window twice on each uplink (unless there's a downlink).

terrillmoore, 2020-01-07:
> how did you come up with 2ms

I looked at the various other implementations and applied "judgement informed by experience". I could easily measure it with my setup, but I wanted to allow for noise, analog settling, etc. I also wanted the receiver (in the 100ppm case) to be spun up before the TX preamble starts, because of laboratory evidence that it helps. That could just be an artifact of the physical design of the Murata module, etc. I believe I also have to leave time for powering up the TCXO on the Murata module. Fundamentally, I suspect 8ms is closer to physical reality for low-noise high gain analog front ends than a few hundred microseconds; but I'm old, and maybe things are much better than I expect.

terrillmoore, 2020-01-07:
The final thing that needs to be done to make the LMIC pass full certification is to get FSK working properly. Perhaps your tool will do that. I need to find a BluePill, or port your code to the STM32L0. @manuelbl does the bluepill's CPU have multiple SPI busses? We could use a second SPI port in parallel and capture the input traffic from the SX1276.  (I'm already scheming on adapting your tool to help with the bringup of the SX1262 port of the LMIC.)

manuelbl, 2020-01-08:
Yes, the BluePill has two SPI buses.

The code should work with any STM32F103C based board. In fact, I have been using three different ones.

The effort to port it to the STM32L0 is probably bigger as it would use a different set of pins and buses and needs other clocks and interrupts.

terrillmoore, 2020-01-08:
Well, I ordered a couple of BluePill devices from Amazon, we'll see how it goes. Thanks again for this work.

cyberman54, 2020-01-27:
Thanks for this great tool!
I did order a BluePill, too. Will analyze the timings in my ESP32 multitasking application with it.
What result should i expect, using latest MCCI LMIC? The margin values listed in the table above?

PS: Can't believe, that other LORAWAN stacks did such precise timing analyzes, i.e. those stacks found on chinese STM32 chips with AT-command UART interface.

manuelbl, 2020-01-27:
There are two things you can look at in the output:

1. *Consistent RX1 and RX2 start*. For any given data rate (spreading factor), the time of the RX1 starts should remain the same (+/- 100µs) across several transmissions. If not, your code is busy and prevents LMIC for opening the RX window. The same applies to all RX2 starts.

2. *Start and end margins*: If item 1 is not an issue, the start and end margins should be close to the values in the LMIC table above. The code hasn't changed since then. If the values are different, we would have to investigate it.  Certainly, the margins shouldn't be negative.

The current LMIC timing assumes that asymmetric margins are advantageous. So far I have assumed otherwise. It depends on the behavior of the SX127x chip. I'm currently running some additional tests (a STM32 MCU controlling two RFM95W modules (sender and receiver) that are half a breadboard apart...). I want to learn more about it (ramp-up time, behavior if two or three preamble symbols are detected and the time-out occurs etc.). I'm close to having results and might come back with a refined model soon.

The current LMIC timing also uses rather narrow margins, much narrower than the Semtech software. If you want to increase them, use `LMIC_setClockError`.

cyberman54, 2020-01-28:
I set up a probing environment with a STM32 bluepill and a ESP32 HeltecV2 board, running my multitasking paxcounter application with MCCI LMIC (current HEAD) on core1. Smooth join and payload communication with SF7. Below the timing log. I am not fully sure yet how to interpret it, can you help? First peek shows huge correction values, i assume that's not a good result?

```
> --------  Sample 1  --------
>   -61872: TX start
>        0: TX done
>           SF7, 125000 Hz, payload = 23 bytes, airtime = 61696us, ramp-up = 176us
>  4998276: RX1 start
>  5072719: RX1: downlink packet received
>           SF7, 125000 Hz, payload = 33 bytes, airtime = 71936us
>           Start of preamble (calculated): 5000783
>           Margin: start = 4255us
> --------  Sample 2  --------
>   -46509: TX start
>        0: TX done
>           SF7, 125000 Hz, payload = 14 bytes, airtime = 46336us, ramp-up = 173us
>   998036: RX1 start
>  1052123: RX1: downlink packet received
>           SF7, 125000 Hz, payload = 20 bytes, airtime = 51456us
>           Start of preamble (calculated): 1000667
>           Margin: start = 4379us
> --------  Sample 3  --------
>   -46510: TX start
>        0: TX done
>           SF7, 125000 Hz, payload = 14 bytes, airtime = 46336us, ramp-up = 174us
>   998071: RX1 start
>  1006574: RX1 timeout
>           SF7, 125000 Hz, airtime = 8192us, ramp-up = 311us
>           Margin: start = 3666us, end = 430us
>           Correction for optimum RX window: -1618us        
>  1998119: RX2 start
>  2027102: RX2 timeout
>           SF9, 125000 Hz, airtime = 28672us, ramp-up = 311us
>           Margin: start = 9762us, end = 2526us
>           Correction for optimum RX window: -3618us
> --------  Sample 4  --------
>   -46511: TX start
>        0: TX done 
>           SF7, 125000 Hz, payload = 15 bytes, airtime = 46336us, ramp-up = 175us
>   998038: RX1 start
>  1006543: RX1 timeout
>           SF7, 125000 Hz, airtime = 8192us, ramp-up = 313us
>           Margin: start = 3697us, end = 399us
>           Correction for optimum RX window: -1649us
>  1998087: RX2 start
>  2027070: RX2 timeout
>           SF9, 125000 Hz, airtime = 28672us, ramp-up = 311us
>           Margin: start = 9794us, end = 2494us
>           Correction for optimum RX window: -3650us
> --------  Sample 5  --------
>   -46511: TX start
>        0: TX done
>           SF7, 125000 Hz, payload = 15 bytes, airtime = 46336us, ramp-up = 175us
>   998040: RX1 start
>  1006545: RX1 timeout
>           SF7, 125000 Hz, airtime = 8192us, ramp-up = 313us
>           Margin: start = 3695us, end = 401us
>           Correction for optimum RX window: -1647us
>  1998089: RX2 start
>  2027075: RX2 timeout
>           SF9, 125000 Hz, airtime = 28672us, ramp-up = 314us
>           Margin: start = 9789us, end = 2499us
>           Correction for optimum RX window: -3645us
> --------  Sample 6  --------
>   -46510: TX start
>        0: TX done
>           SF7, 125000 Hz, payload = 15 bytes, airtime = 46336us, ramp-up = 174us
>   998048: RX1 start
>  1006551: RX1 timeout
>           SF7, 125000 Hz, airtime = 8192us, ramp-up = 311us
[... 48 lines cut ...]
>   998040: RX1 start
>  1006542: RX1 timeout
>           SF7, 125000 Hz, airtime = 8192us, ramp-up = 310us
>           Margin: start = 3698us, end = 398us
>           Correction for optimum RX window: -1650us
>  1998090: RX2 start
>  2027071: RX2 timeout
>           SF9, 125000 Hz, airtime = 28672us, ramp-up = 309us
>           Margin: start = 9793us, end = 2495us
>           Correction for optimum RX window: -3649us
> --------  Sample 11  --------
>   -46511: TX start
>        0: TX done
>           SF7, 125000 Hz, payload = 15 bytes, airtime = 46336us, ramp-up = 175us
>   998038: RX1 start
>  1006542: RX1 timeout
>           SF7, 125000 Hz, airtime = 8192us, ramp-up = 312us
>           Margin: start = 3698us, end = 398us
>           Correction for optimum RX window: -1650us
>  1998089: RX2 start
>  2027072: RX2 timeout
>           SF9, 125000 Hz, airtime = 28672us, ramp-up = 311us
>           Margin: start = 9792us, end = 2496us
>           Correction for optimum RX window: -3648us
```

terrillmoore, 2020-01-28:
@cyberman54 -- it depends on what you mean by "optimum". This shows that the LMIC is starting earlier than @manuelbl's code expects. His code's expectation is based on Semtech's published recommendations; LMIC is tuned for best likelihood of reception based on empirical tests. It definitely uses more battery power in the case of lots of downloads at low data rates (high spreading factors); but in my testing it has better PER. The window timing (for battery power) only has significant effect at low data rates, because the preamble is very long, and you must have the recevier on for longer.  But if you're not downlinking a lot, there is *no* difference between the LMIC timing and the Semtech reference code timing; both listen for a certain number of symbols and then time out. Same number of symbols ==> same power.

Furthermore, receiver-on power is roughly 10% of the transmitter-on power. Since there's at least one transmit per receive, the timing of the RX window has at most a 2nd-order (10%) effect on battery life; and probably much less, because the preamble time is only a small fraction of the overall receive message time.

cyberman54, 2020-01-28:
@terrillmoore thanks for the explanation. I'm aware that RX timing is not the big battery drainer, but that's not the reason why i am trying to get a closer look on timing. I'm still struggling with RX problems in my application, like (too) many join retries and sometimes lost downlink payload. My goal is to make sure, that there are no LMIC timing issues caused by running the LMIC stack in my multitasking ESP32 environment.

cyberman54, 2020-01-28:
Some more samples for SF12:

```
--------  Sample 41  --------
-1155266: TX start
       0: TX done
          SF12, 125000 Hz, payload = 15 bytes, airtime = 1155072us, ramp-up = 194us
  998048: RX1 start
 1227741: RX1 timeout
          SF12, 125000 Hz, airtime = 229376us, ramp-up = 317us
          Margin: start = 67171us, end = 31133us
          Correction for optimum RX window: -18019us
 1998099: RX2 start
 2027082: RX2 timeout
          SF9, 125000 Hz, airtime = 28672us, ramp-up = 311us
          Margin: start = 9782us, end = 2506us
          Correction for optimum RX window: -3638us
--------  Sample 42  --------
-1155266: TX start
       0: TX done
          SF12, 125000 Hz, payload = 15 bytes, airtime = 1155072us, ramp-up = 194us
  998374: RX1 start
 1228067: RX1 timeout
          SF12, 125000 Hz, airtime = 229376us, ramp-up = 317us
          Margin: start = 66845us, end = 31459us
          Correction for optimum RX window: -17693us
 1998421: RX2 start
 2027407: RX2 timeout
          SF9, 125000 Hz, airtime = 28672us, ramp-up = 314us
          Margin: start = 9457us, end = 2831us
          Correction for optimum RX window: -3313us
--------  Sample 43  --------
-1155268: TX start
       0: TX done
          SF12, 125000 Hz, payload = 15 bytes, airtime = 1155072us, ramp-up = 196us
  998041: RX1 start
 1227733: RX1 timeout
          SF12, 125000 Hz, airtime = 229376us, ramp-up = 316us
          Margin: start = 67179us, end = 31125us
          Correction for optimum RX window: -18027us
 1998092: RX2 start
 2027074: RX2 timeout
          SF9, 125000 Hz, airtime = 28672us, ramp-up = 310us
          Margin: start = 9790us, end = 2498us
          Correction for optimum RX window: -3646us

```

terrillmoore, 2020-01-28:
@cyberman54 sorry for being so pedantic! (I was really writing for the people who find this later with web search.)

It strikes me that you should look at the received packet RSSI and SNR. I fixed the calculations for that in recent builds, so I think they're more meaningful. It also strikes me that you could use SPI or a serial port to copy the credentials to a second device, using something like the rwc_nst_test sketch, but close to your main device -- and then you could display downlinks independent of timing. You'd basically have to get the downlink channel to the second device in time, and trigger a continuous read at the approprate spreading factor, etc. Indeed, it strikes me that we could probably hack your device to do a continous read rather than a windowed read, for test purposes. But the thing to do would be side-by-side testing, one windowed, one continuous, and see if the data shows up at all. If it's not on the air, we're not going to receive it.

terrillmoore, 2020-02-09:
At the Things Conference last month, I heard that LimeSDR can be used for LoRa packet sniffing. I don't know if it's true or not, but this might be the best way to find out whether this is an LMIC problem or a "no data" problem. In any case, @cyberman54, how should we proceed with this ticket?

manuelbl, 2020-02-09:
I'm still working on a more refined timing model and trying to figure out
how much time the ramp-up take, how much time is spent in post-processing
the received packet until the interrupt is triggered and if the timeout is
can be extended and still raise the timeout interrupt.

I hope to have some results within a week (if I can sort out the difference
I currently see between two setups).

Can you leave it open for a few more days?

terrillmoore, 2020-02-09:
Of course, happy to. Just doing a sweep on things and pinging people.

cyberman54, 2020-02-10:
> At the Things Conference last month, I heard that LimeSDR can be used for LoRa packet sniffing.

Thanks for this hint. I will order a Lime Mini SDR to test.

cyberman54, 2020-02-12:
I got a Lime Mini SDR. Any hints / suggestions what software to use for Lora timing analysis?

manuelbl, 2021-02-23:
I'm unlikely to find time to continue with the analysis. Therefore, I'm closing it.

closed 2021-02-23 by manuelbl