mcci-catena/arduino-lmic #522: Upload messages sent but lost

DeveloppeurPascal, 2020-01-10:
I've a strange behavior : sometimes the payload I sent don't arrive on the gateways.

In this case the first 3 messages are lost, the 2 lasts messages are received and appear on Orange LiveObjects.

I don't understand why. Do you see any explanation or have suggestion to solve this ?

I use an Heltec Wifi Lora 32 v2 in Europe (868Mhz).

=> join a gateway in 1 try

151099: LoRa: EV_JOINING
152887: engineUpdate, opmode=0x4
203181: engineUpdate, opmode=0x4
203206: LoRa: EV_TXSTART
205205: TXMODE, freq=868100000, len=23, SF=7, BW=125, CR=4/5, IH=0
520821: setupRx1 txrxFlags 00 --> 01
start single rx: now-rxtime: 3
521451: RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 521980 rxtime: 521445 entry-rxtime: 535 now-entry: 3 rxtime-txend: 312375
583320: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 3
583951: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
698916: Setup channel, idx=3, freq=866700000
698925: Setup channel, idx=4, freq=867100000
698929: Setup channel, idx=5, freq=867300000
699543: Setup channel, idx=6, freq=867700000
702538: Setup channel, idx=7, freq=867900000
705540: LoRa: EV_JOINED
709367: engineUpdate, opmode=0x800

=> next Payload not received by Orange Live Objects

15492791: engineUpdate, opmode=0x808
15492842: LoRa: EV_TXSTART
15494799: TXMODE, freq=868100000, len=61, SF=7, BW=125, CR=4/5, IH=0
15813614: setupRx1 txrxFlags 0x2 --> 01
start single rx: now-rxtime: 3
15814244: RXMODE_SINGLE, freq=868100000, SF=9, BW=125, CR=4/5, IH=0
rxtimeout: entry: 15816053 rxtime: 15814238 entry-rxtime: 1815 now-entry: 3 rxtime-txend: 312375
15876114: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 3
15876744: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 15891096 rxtime: 15876738 entry-rxtime: 14358 now-entry: 4 rxtime-txend: 374875
15891114: processRx2DnData txrxFlags 0x2 --> 00
15892375: processDnData_norx txrxFlags 00 --> 20
15895629: LoRa: EV_TXCOMPLETE (includes waiting for RX windows)
15901761: engineUpdate, opmode=0x900

=> next Payload not received by Orange Live Objects

19657580: engineUpdate, opmode=0x908
19657628: LoRa: EV_TXSTART
19659730: TXMODE, freq=868300000, len=61, SF=7, BW=125, CR=4/5, IH=0
19978545: setupRx1 txrxFlags 0x20 --> 01
start single rx: now-rxtime: 3
19979176: RXMODE_SINGLE, freq=868300000, SF=9, BW=125, CR=4/5, IH=0
rxtimeout: entry: 19980985 rxtime: 19979170 entry-rxtime: 1815 now-entry: 4 rxtime-txend: 312375
20041046: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 3
20041676: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 20056029 rxtime: 20041670 entry-rxtime: 14359 now-entry: 3 rxtime-txend: 374875
20056046: processRx2DnData txrxFlags 0x2 --> 00
20057307: processDnData_norx txrxFlags 00 --> 20
20060562: LoRa: EV_TXCOMPLETE (includes waiting for RX windows)
20066693: engineUpdate, opmode=0x900

=> next Payload not received by Orange Live Objects

26377566: engineUpdate, opmode=0x908
26377619: LoRa: EV_TXSTART
26379687: TXMODE, freq=868500000, len=62, SF=7, BW=125, CR=4/5, IH=0
26698823: setupRx1 txrxFlags 0x20 --> 01
start single rx: now-rxtime: 3
26699453: RXMODE_SINGLE, freq=868500000, SF=9, BW=125, CR=4/5, IH=0
rxtimeout: entry: 26701262 rxtime: 26699447 entry-rxtime: 1815 now-entry: 4 rxtime-txend: 312375
26761323: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 3
26761953: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 26776305 rxtime: 26761947 entry-rxtime: 14358 now-entry: 4 rxtime-txend: 374875
26776322: processRx2DnData txrxFlags 0x2 --> 00
26777590: processDnData_norx txrxFlags 00 --> 20
26780838: LoRa: EV_TXCOMPLETE (includes waiting for RX windows)
26786986: engineUpdate, opmode=0x900

=> next Payload received by Orange Live Objects (with Fcnt=3)

34874410: engineUpdate, opmode=0x908
34874463: LoRa: EV_TXSTART
34876523: TXMODE, freq=866700000, len=62, SF=7, BW=125, CR=4/5, IH=0
35195658: setupRx1 txrxFlags 0x20 --> 01
start single rx: now-rxtime: 3
35196289: RXMODE_SINGLE, freq=866700000, SF=9, BW=125, CR=4/5, IH=0
rxtimeout: entry: 35198098 rxtime: 35196283 entry-rxtime: 1815 now-entry: 4 rxtime-txend: 312375
35258158: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 3
35258789: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 35273141 rxtime: 35258783 entry-rxtime: 14358 now-entry: 4 rxtime-txend: 374875
35273160: processRx2DnData txrxFlags 0x2 --> 00
35274421: processDnData_norx txrxFlags 00 --> 20
35277675: LoRa: EV_TXCOMPLETE (includes waiting for RX windows)
35283832: engineUpdate, opmode=0x900

=> next Payload received by Orange Live Objects (with Fcnt=4)

52871565: engineUpdate, opmode=0x908
52871618: LoRa: EV_TXSTART
52873570: TXMODE, freq=867100000, len=62, SF=7, BW=125, CR=4/5, IH=0
53192705: setupRx1 txrxFlags 0x20 --> 01
start single rx: now-rxtime: 3
53193335: RXMODE_SINGLE, freq=867100000, SF=9, BW=125, CR=4/5, IH=0
rxtimeout: entry: 53195143 rxtime: 53193329 entry-rxtime: 1814 now-entry: 4 rxtime-txend: 312375
53255205: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 3
53255835: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 53270188 rxtime: 53255829 entry-rxtime: 14359 now-entry: 3 rxtime-txend: 374875
53270206: processRx2DnData txrxFlags 0x2 --> 00
53271467: processDnData_norx txrxFlags 00 --> 20
53274721: LoRa: EV_TXCOMPLETE (includes waiting for RX windows)
53280878: engineUpdate, opmode=0x900

terrillmoore, 2020-01-11:
Sorry you're having problems. The log shows that the messages are begin sent. You don't mention your hardware or antenna, but the symptoms point to traffic loss in the network, not an LMIC problem.

DeveloppeurPascal, 2020-01-11:
It's what I thought. Thanks for confirming.

terrillmoore, 2020-02-09:
@DeveloppeurPascal may I close this?

DeveloppeurPascal, 2020-02-18:
Hi, I have to do some tests today and will close if all is ok.

DeveloppeurPascal, 2020-02-18:
It finally seems to be ok with last release and your sample code. Thanks for your work and help.

closed 2020-02-18 by DeveloppeurPascal

terrillmoore, 2020-02-18:
Great, thanks!