mcci-catena/arduino-lmic #525: ttn-otaa-dragino-lorashield-915 LMIC.frequency set to 0

aw5m2nite, 2020-01-15:
Hi, 

I've been trying out the example code "ttn-otaa-dragino-lorashield-915".  The code compiles, but in the Serial Monitor, it is saying that it is "Sending packet on frequency: 0".   

I'm not seeing the device on my TTN applicaiton with OTAA, but have tested the "ttn-apb-dragino-lorashield-915", and that is able to connect to TTN and I am seeing data in my TTN application.   I have added in the device's App EUI, Device EUI and App Key.  
[image]

terrillmoore, 2020-01-15:
Sorry you're having problems.

Those examples are not part of this project. Are you sure that you are filing the report in the right place?

terrillmoore, 2020-02-09:
Hi @aw5m2nite, if there's no further discussion, may I close this?

caueb, 2020-09-23:
I was having the same issue and I found out that my Single Channel gateway was not receiving the data. When I tried with a multiple channel gateway it works.

closed 2020-12-10 by terrillmoore