mcci-catena/arduino-lmic #526: removeApbChangeCallback() fail while initializing LMIC on ESP32 system

cyberman54, 2020-01-26:
With current version of arduino-esp32 a new moduel `esp32-hal-cpu.c` was introduced, which improved error handling in context with CPU clock switching / management.

If i run my LMIC application on top of this, there is an error showing something goes wrong with removing an application callback handler.

I am not sure yet if this problem ist caused by either LMIC, my application, or the new  `esp32-hal-cpu.c`. I wrote a comment to this[ issue](https://github.com/espressif/arduino-esp32/pull/3560), maybe someone with deep LMIC knowledge could peek into it to get a clue what's going on. Thanks.

terrillmoore, 2020-01-26:
See my comment on https://github.com/espressif/arduino-esp32/pull/3560. I don't think this is an LMIC problem.

cyberman54, 2020-01-26:
Solved. This was an arduino-esp32 related problem.

closed 2020-01-26 by cyberman54