mcci-catena/arduino-lmic #533: Low power example

jpmeijers, 2020-02-07:
In the past I have been struggling to make LMiC work together with my microcontroller going into deep sleep mode. Currently I do a call to `os_queryTimeCriticalJobs(ms2osticks(5000))` after which I will go into sleep mode for 5 seconds.

One remaining issue is that the micros() timer does not increment in sleep mode, so LMiC will think no time has passed, and the duty cycle logic will prevent me from sending messages - something that I want to do every 10 minutes.

Would it be possible to provide a minimal example of how one can use sleep modes along with LMiC?

Depending on the platform I either use RTCZero and schedule alarms 5 seconds after going to sleep, or I use the WDT to wake up every 8 seconds.

JackGruber, 2020-02-07:
Hi,

a example from my project. Sleeptime is 8 second.

**Update timer**
```
// LMIC uses micros() to keep track of the duty cycle, so
// hack timer0_overflow for a rude adjustment: 
cli();
timer0_overflow_count+= 8 * 64 * clockCyclesPerMicrosecond();
sei();
```

terrillmoore, 2020-02-09:
Yep, updating the clock after sleep is something MCCI does in our STM32 and SAMD sketches. This was the original reason we forked BSPs, as there was no good way to do this in the stock BSPs. (Then we realized it was easy to configure the LMIC, and all our projects are using the LMIC, so...)

mchacher, 2020-04-06:
Hi all,
I am very much interested in this thread having exactly the same question. I am not able to understand how to manage the deep sleep mode with an Arduino Pro mini. Could you please share a more detailed example?
thanks a lot.

JackGruber, 2020-04-06:
Hi, you can take a look in my project.
https://github.com/JackGruber/Arduino-Pro-Mini-LoRa-Sensor-Node

mchacher, 2020-04-06:
Thanks for sharing @JackGruber. I will look at your code.

jaqPi, 2020-08-05:
Hello everyone,

I run into the same problem. Thanks @JackGruber for sharing your project and your solution. Unfortunately, the code works only for AVR boards but not for my Feather M0. Do you know how do adjust the millis() timer of a Feather M0? I am about to read myself into that topic, but it still feels like black magic at the moment.

etix, 2021-06-01:
> Yep, updating the clock after sleep is something MCCI does in our STM32 and SAMD sketches. This was the original reason we forked BSPs, as there was no good way to do this in the stock BSPs. (Then we realized it was easy to configure the LMIC, and all our projects are using the LMIC, so...)

I've spent quite a few hours trying to make my SAMD board sleep while keeping track of the elapsed time waiting for the RX windows (especially now since TTN v3 requires 5 seconds), I tried to manually adjust os_gettime but it failed since the MCU can take an unpredictable number of ticks waking up and makes the LMIC unhappy.  @terrillmoore would you mind sharing how you did it?