mcci-catena/arduino-lmic #534: downlink queue

geppoleppo, 2020-02-09:
First of all thanks for the huge job!
I ask for clarification regarding the messages in downlink.
If with the old library (vers. 2.3.2) I put several messages in a queue and they were downloaded in sequence, with the new library it is downloaded one at a time and therefore if, for example, I put in the queue 5 messages 5 different uplinks are needed to empty the queue.
Is there any parameter to set?
Thanks

terrillmoore, 2020-02-09:
Sorry you're having troubles. Don't know why this would have stopped working; the code that handles this is unchanged.

https://github.com/mcci-catena/arduino-lmic/blob/79accbbfb9d700604a6915b2a6937578f1983c5a/src/lmic/lmic.c#L1306-L1307

This code says:  if this was a confirmed downlink, or if the `FPending` bit was set, initiate a dummy uplink; this will ensure that the ack for the downlink gets sent, or that the network has another downlink. The poll will take place as soon as the current RX cycle completes (which is right away, because we're processing a valid downlink, which ends the RX cycle).

Can you answer a few questions:

- What hardware are you using?
- Which network are you using, and which region?
- Is the network setting the `FPending` (which the LMIC calls `FCT_MORE` bit in the downlink?

geppoleppo, 2020-02-09:
> * What hardware are you using?
MCU: ATMEGA 1284p
Lora chip: hope rfw95

> * Which network are you using, and which region?
Resiot Loraserver (https://www.resiot.io/)-Europe region

> * Is the network setting the `FPending` (which the LMIC calls `FCT_MORE` bit in the downlink?
I'm not sure but I think the Fpending is set.
With the old library the messages were downloaded in sequence while with the new one not.
For security, I ask the Iot platform manager.

terrillmoore, 2020-02-10:
There are two possibilities. Either it's boken, or a change in the LMIC causes the network to behave likely. I suspect hte later. I'll see if I can get someone from MCCI to test the FPending bit; but because the current LMIC is much better about setting data rates according to the LoRaWAN standard, it seems most likely that you're running at a different (slower) data rate. This will result in the gateway not being able to send downlinks as rapidly as before, because of duty cycle limits on the downlink channels in EU. The LMIC doesn't take into account the downlink duty cycle in sending its uplink polls.

You could try manually setting the datarate to EU868_SF7 (DR5) after you join, and then see how things go.

Your network provider probably has a console for looking at uplinks/downlinks. If you can find that and send a trace, I'll be able to guess at what might be happening.