mcci-catena/arduino-lmic #539: PlatformIO travis builds are broken

terrillmoore, 2020-02-14:
As of a week or so ago, Travis CI builds are broken. I don't think it was correlated with the 3.1.0 release, but... it's possible.

The broken builds are for ESP32 and AVR.

I don't know enough about PlatformIO to fix this. The broken builds are: 

- https://travis-ci.com/mcci-catena/arduino-lmic/jobs/286942171
- https://travis-ci.com/mcci-catena/arduino-lmic/jobs/286942172

@frazar, @manuelbl, any ideas as to what might be wrong?

frazar, 2020-02-14:
I can reproduce the error on my machine. I believe it is related to PlatformIO version 4.2, since the bug disappears when using 4.1.

frazar, 2020-02-14:
I think it is caused by platformio/platformio-core#3377.
A fix has already been prepared, and will be released as part of PlatformIO 4.2.1.

terrillmoore, 2020-02-14:
Thanks @frazar!

terrillmoore, 2020-02-17:
Fixed by https://github.com/mcci-catena/arduino-lmic/commit/87c9f2be531671a7f02959dcfc31b816340ca6b0/ (#543)

closed 2020-02-17 by terrillmoore