mcci-catena/arduino-lmic #541: Lmic returns unknown Event Codes 100 and 20

dardrai, 2020-02-15:
**Unchatched events**
in the methode void onEvent(ev_t event) some event code are getting dispatched and I don't know what they are for the message codes are 100 and 20

**Environment**

- Version of LMIC being used: 3.0.99 (but it also earlier lib version had this "issue")
- Version of Arduino IDE being used: 1.8.11
- Network provider: TTN
- Region:EU
- Board: NRF52840 and Feather M4 Express
- Radio: Adafruit LoRa Radio FeatherWing - RFM95W (SX127x LoRa)

Example Message Code 100:
https://i.imgur.com/jBBkkQL.png

Normaly (if Debug is activated) the output should look like:
12:22:05.651 -> EV_TXSTART
12:22:05.651 -> Message Code: 17
12:22:10.817 -> EV_JOINED
12:22:10.817 -> Message Code: 6
12:22:10.817 -> EV_TXSTART
12:22:10.817 -> Message Code: 17
12:22:12.912 -> EV_TXCOMPLETE

I'm shure that the OTAA Keys are stored in the right format and values (not like in Issue #492 ) then sometimes I receive the Event Code 100 or 20 - and later the lmic "recovers" and is sending correctly.
What is the meaning of these two event Codes?

dardrai, 2020-02-15:
ConsoleLog (only Event Code 100) - Event Code 20 is very rare (could it be that this is Code is only thrown if something did happen on the TTN Side?)

12:45:48.362 -> EV_TXSTART
12:45:48.362 -> Message Code: 17
**12:45:48.362 -> Message Code: 100**
12:45:50.465 -> EV_TXCOMPLETE
12:46:20.474 -> Message Code: 10
12:46:20.474 -> Volt: 3.74
12:46:20.474 -> Percent: 43
12:46:20.474 -> Range: 265
12:46:20.474 -> EV_TXSTART
12:46:20.474 -> Message Code: 17
**12:46:20.474 -> Message Code: 100**
12:46:22.576 -> EV_TXCOMPLETE
12:46:52.598 -> Message Code: 10
12:46:52.598 -> Volt: 3.74
12:46:52.598 -> Percent: 43
12:46:52.598 -> Range: 265

terrillmoore, 2020-02-15:
See comment on #542. The text "Message code" appears nowhere in the arduino-lmic liibrary, so you are running a sketch that I'm not aware of. I have no idea what that might be.

dardrai, 2020-02-16:
the text "Message Code:" I'll add in my sketch for every event (when the debug is on) this is just for debug/dev purpose. But the event with this number is thrown somewhere from lmic/ttn or gw

In your Example you'd catch these errors with default case:
default:
Serial.print(F("Unknown event: "));
Serial.println((unsigned) ev);
break;

and then the out would be:
Unknown event: 100
Unknown event: 20

I found in the file src\lmic\lmic.h fallowing enum that could match the error code:
enum { MAX_MISSED_BCNS = 20 }; // threshold for triggering rejoin requests
enum { MAX_RXSYMS = 100 }; // stop tracking beacon beyond this

terrillmoore, 2020-02-18:
@dardrai -- those are not error codes. In addition, we don't see those errors in other BSPs, so something else is going on. 

I don't have your hardware or your sketch, so I'm unable to test; I'm happy to advise or correct the problem, but need your help to find it. My suggestion is that you use `git bisect` to find the commit that broke your code. If you're not familiar with using `git bisect` it's straighforward:

1. install git (if needed)
2. move the arduino-lmic from your arduino libraries directory to some other place.
3. use `git clone` to checkout arduino-lmic from github
4. `cd` to the arduino-lmic directory.
5. `git checkout v3.1.0`
6. `git bisect start`
7. `git bisect bad` (because v3.1.0 is bad)
8. `git bisect good v3.99.3` (because v3.99.3 is ok)
9. Git has now checked out a revision. Compile and test.
10. If this rev is good say `git bisect good`; otherwise `git bisect bad`.
11. repeat steps 9 and 10 until git says "no commits left"; it will leave you at the bad commit.
12. use `git log --oneline` to show the commit history from the bad commit down, and post that here.

dardrai, 2020-02-21:
Okay I did found the problem - Error Code 20 is EV_JOIN_TXCOMPLETE which was missing from my sketch.

closed 2020-02-21 by dardrai