mcci-catena/arduino-lmic #542: Lmic won't send Data after Update

dardrai, 2020-02-15:
I have a sketch that is working with the lmic Version 3.0.99 perfectly fine. 
 Console out:
12:41:58.372 -> EV_TXSTART
12:41:58.372 -> Message Code: 17
12:42:03.500 -> EV_JOINED
12:42:03.500 -> Message Code: 6
12:42:03.500 -> EV_TXSTART
12:42:03.500 -> Message Code: 17
12:42:05.608 -> EV_TXCOMPLETE

as soon the lmic Version is updated to 3.1 fallowing function won't work anymore:
- Upgrade Sketch, the USB Port is "blocked" if I want to upgrade the running sketch I'll have to enter the DFU Mode manually on the NRF52
- Console Output is:
12:41:58.372 -> EV_TXSTART
12:41:58.372 -> Message Code: 17
and nothing Happen after that - in the TTN Portal, I just see Auth request but not more ==> e.g. EV_JOINED is never shown in the console

**Environment**
    Version of LMIC being used: 3.1
    Version of Arduino IDE being used: 1.8.11
    Network provider: TTN
    Region:EU
    Board: NRF52840 and Feather M4 Express
    Radio: Adafruit LoRa Radio FeatherWing - RFM95W (SX127x LoRa)

**To Reproduce**
Run a version from Issue #535 in 3.0.99 Working
Upgrade Lmic to 3.1 => Sketch not Running anymore

**From old Comment**
Tried the new Version 3.1.0 again - but the NRF doesn't work well with new Version?

    The Sketch above ("Main Class") does make an auth request, but never sends (this sketch works fine on 3.0.99)
    After programing the NRF with 3.1.0 the Serial Port (over USB) doesn't work as expected

    ArduinoIDE can't open the Serial Monitor if the NRF is running
    If the NRF receive a reset (via button) the Serial Monitor opens but it will be empty
    Update the running sketch nearly impossible with the lib 3.1.0 (*I believe it has something to do with the overall Serial USB Connection) | to update the NRF, the NRF has the be put in DFU Mode

After reverting back to Version 3.0.99 and downgrading the running sketch everything works as expected - NRF goes Online and send his data, Device can be programed and watch via the serial Monitor etc.

Update to the USB Serial Terminal - If the Terminal is Open the Output is shown, but if the Terminal hadn't been open then then the Problem is still occurring (reboot the NRF, so that the ArduinoIDE can open the Terminal)

[image]

Tried to add the Fix: v3.1.0 - sx1272 - line misspelled to the lib but this didn't change much

terrillmoore, 2020-02-15:
Sorry, but the text you're showing is not from any of our sample sketches (the text "Message code" does not appear anywhere in the source base).  I suggest you use an *unchanged* sketch from the examples, for example the compliance sketch, and see if it works.

I have no idea what is going on or what those "message codes" are or what they might mean. I suggest that you use git bisect to find the commit that broke things for you. I suspect that #523 etc. are the root cause, but as I don't have your hardware and things are working fine for me....

dardrai, 2020-02-15:
the text "Message Code:" I'll add in my sketch for every event (when the debug is on) this is just for debug/dev purpose. But the event with this number is thrown somewhere from lmic/ttn or gw 

In your Example you'd catch these errors with default case:
  default:
            Serial.print(F("Unknown event: "));
            Serial.println((unsigned) ev);
            break;

and then the out would be:
Unknown event: 100
Unknown event: 20

I found in the file src\lmic\lmic.h fallowing enum that could match the error code:
enum { MAX_MISSED_BCNS    =  20 };   // threshold for triggering rejoin requests
enum { MAX_RXSYMS         = 100 };   // stop tracking beacon beyond this

dardrai, 2020-02-16:
I'll test an example and will give a feedback  - but the msg code is just a log output for me for developping. The codes that i'll print are the real value behind e.g EV_TXSTART == 17.

czuvich, 2020-02-17:
I can confirm this issue on the nRF52 (Adafruit nRF52840 Express) + RFM95.  Using the 3.1.0 causes an issue while 3.0.99 does not.  Here's my synopsis:

1) Use the ttn-otaa-feather-us915 sketch and configure the lmic_pins specific to the nRF52840 configuration (remove the other defines).
2) Disconnect a local gateway so that the device cannot JOIN (I'll explain why later).
2) Run the sketch.

With 3.1.0 - Two events come up into the console (EV_JOINING followed by EV_TXSTART); however, the sketch just stops outputting anything else after a short amount of time.
I put a couple of Serial.println() with some counter in the loop() function, and after about 6-7 seconds it stops printing the counter variable (the Sketch freezes).

With 3.0.99 - I see a consistent EV_TXSTART and EV_JOIN_TXCOMPLETE: no JoinAccept which is what I'd expect.

Now, here's what I think is happening.  I think there's a memory issue in the os_runloop_once() that has been introduced.  If I put delays before the os_runloop_once(), it takes a little longer to freeze.

The reason I wanted to keep from connecting to a gateway was to isolate the issue.  Thoughts?

dardrai, 2020-02-22:
Thank you @czuvich for testing it with the example and reporting same error on nearly identical hardware.
Too test it further and maybe eliminate hardware that isnt't the problem of this bug/error, I testet with a different SOC - the Adafruit Feather M4 Express + RFM95 Module.

The ttn-otaa (example) script is running as it should with the Version 3.0.99 but as soon I Upgarde the Lib to 3.1 the lib function break again. The behaviour is nearly identical as with the nrf52
USB Serial Connection is Buggy and won't open, if the Sketch is already running
If I restart the M4 via button, the Console open up and just report EV_JOINING followed by EV_TXSTART - but not more. 
Only Auth request are visible on the TTN Console - but no Uplink Data:
[image]

Do to this behaviour I don't think it is a problem with the NRF52 or the M4 Express (they are completely different - e.g. NRF52 uses FreeRTOS and the M4 clearly does not) - only the RFM95 Module is the same in both version - so I think that the Problem should be there.

terrillmoore, 2020-02-22:
@dardrai : I have previously stated in #541: I do not have *any* of the above hardware. I cannot repro or debug your problem. Please review my instructions on how to use git bisect, or find a developer with the hardware and the git confidence to do this and repro the problem. Once you find the commit that broke things, I can fix. Until this is done, I can do nothing.

For your reference, this is what I wrote to you about git bisect in #541:

> I don't have your hardware or your sketch, so I'm unable to test; I'm happy to advise or correct the problem, but need your help to find it. My suggestion is that you use `git bisect` to find the commit that broke your code. If you're not familiar with using `git bisect` it's straighforward:
> 
>     1. install git (if needed)
> 
>     2. move the arduino-lmic from your arduino libraries directory to some other place.
> 
>     3. use `git clone` to checkout arduino-lmic from github
> 
>     4. `cd` to the arduino-lmic directory.
> 
>     5. `git checkout v3.1.0`
> 
>     6. `git bisect start`
> 
>     7. `git bisect bad` (because v3.1.0 is bad)
> 
>     8. `git bisect good v3.99.3` (because v3.99.3 is ok)
> 
>     9. Git has now checked out a revision. Compile and test.
> 
>     10. If this rev is good say `git bisect good`; otherwise `git bisect bad`.
> 
>     11. repeat steps 9 and 10 until git says "no commits left"; it will leave you at the bad commit.
> 
>     12. use `git log --oneline` to show the commit history from the bad commit down, and post that here.

czuvich, 2020-02-26:
FYI, for some reason I can’t use OTAA even with 3.0.99 now.  I can see the Join Accept on the Gateway but the node keeps saying No Join Accept. ABO works.  I’ll open another ticket with logs.

@dardrai if you could run through bisect that’d be awesome. I tried doing it last night but ran into the above. The steps he posted are exactly what you need to do except you need to put “3.0.99.3” not “3.99.3”.

terrillmoore, 2020-12-10:
I think that this can be closed now. Reopen it if needed (and possibly convert to a discussion).

closed 2020-12-10 by terrillmoore