mcci-catena/arduino-lmic #545: Payload disappear on Orange Live Object

DeveloppeurPascal, 2020-02-19:
**Describe your question or issue**

Hi

Finally I retried the sample program for OTAA on Orange LiveObject and I have the same problem I have on my projects : Payload are not sent or disappear before getting received by Orange.

I modified the sample program to send payloads in a loop and increase the first byte (to have a counter) each time I send something.

I don't know if I do something wrong by using the library or if the problem is between the library and Orange LoRaWan gateway. For them all is okay on their network but they changed something recently : they send MAC requests and the library answer to it automatically as LoRaWan protocol needs it.

Except this problem of ghost payload, I have an other one : sometimes the JOIN process don't work and need a lot of try, but on Orange LiveObject the log say it accepted the JOIN. The library stop with TX_JOIN_COMPLETE and restart a join exchange.

I'm not so far from gateway antenna and tried on other places with other Orange gateways.

If you have any idea how to fix it or what I can change, I'll appreciate very much.

**Environment**

This information is all important; it's hard to help without a complete set of answers.

- Version of LMIC being used: last one, downloaded from GitHub this week
- Version of Arduino IDE being used: 1.8.10
- Network provider (The Things Network, Swisscom, ChirpStack, etc.) Orange (in France)
- Region (EU868, US915, etc.) EU868
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.) Heltec Wi-Fi LoRa 32 v2
- Radio (HopeRF, SX1276, etc.) SX1276

[data_19_02_2020_210813.txt](https://github.com/mcci-catena/arduino-lmic/files/4227250/data_19_02_2020_210813.txt)
[library-log.txt](https://github.com/mcci-catena/arduino-lmic/files/4227257/library-log.txt)
[modified-otaa-sample.txt](https://github.com/mcci-catena/arduino-lmic/files/4227261/modified-otaa-sample.txt)

cyberman54, 2020-03-01:
Perhaps this is an RX1DRoffset issue. LMIC is using RX1DRoffset=0 by default. Are you sure that this setting matches requirements of Orange Lora network? Some commercial Lora network operators use RX1DRoffset=2 (e.g. ThingPark) or 3 (e.g. KPN), as far as i understood from postings on the internet.

terrillmoore, 2020-03-01:
@DeveloppeurPascal If you can send me a summary of the device the requirements for the Orange network for OTAA, I'll take a look.

DeveloppeurPascal, 2020-03-04:
I found the reason of the problem : automatic MAC response throw TXCOMPLETE event, so I send next message, but the previous payload was pending... a new payload is sent during send process from the previous one, so it disappear.

We must modify the do_send() function like this :

  if (LMIC.opmode & OP_TXRXPEND) {
   Serial.println(F("OP_TXRXPEND, not sending"));
  } else if (LMIC.opmode & OP_TXDATA) {
   Serial.println(F("OP_TXDATA, not sending"));
  } else {
      LMIC_setTxData2(1, LoRaWanMsgToSend, LoRaWanSizeOfMsgToSend, 0);
      // Next TX is scheduled after TX_COMPLETE event.
  }

Please add the OP_TXDATA test in your samples to prevent the problems I add.

Thanks

terrillmoore, 2020-03-04:
Yes, you're right. My apologies for the inconvenience.

The behavior of the LMIC changed and the sample scripts didn't adapt. More precisely, the LMIC is now rejecting attempts to send while busy; before, it didn't... and the LMIC now correctly handles a range of MAC commands that it previously did not, so it's more likely to be busy.

Unfortunately, all of us who were testing, were testing with actual apps rather than the sample apps. Several of us ran into that problem, but we fixed our apps. I'll open a new ticket for that, but close this one.

closed 2020-03-04 by terrillmoore