mcci-catena/arduino-lmic #548: 3.1.0 breaks example `ttn-otaa-feather-us915` - M0 stuck?

pomplesiegel, 2020-03-03:
**tl;dr: The example works on a Feather M0 900Mhz with v2.3.2, but not with v3.1.0.** 

First: Huge fan of this project and eternally grateful for everyone's hard work. Thank you very much. 

Quite excited about v3.1.0, but this release (and tags before it + code pushes after it) don't seem to be working on my custom M0 board nor a Feather M0 i just took out of the box. 

I'm wondering if this is related to issue #542 but I am trying not to make any assumptions. 

To get back to basics and isolate the issue, I am just running the "ttn-otaa-feather-us915.ino" example, adding in only my keys on a Feather M0 LORA 900Mhz. 

Raw captures from both sw versions:

**v3.1.0**

Here is the output and a picture of TTN console
Both the serial logs and the TTN output get stuck here indefinitely:
````
12:24:46.570 -> Starting
12:24:46.570 -> Packet queued
12:24:46.570 -> 313343: EV_JOINING
12:24:46.570 -> 313419: EV_TXSTART
````
[image]

**v2.3.2**

In contrast, things work as expected with this existing version

````
12:22:58.235 -> Starting
12:22:58.272 -> Packet queued
12:22:58.272 -> 313352: EV_JOINING
12:22:58.272 -> 313430: EV_TXSTART
12:23:03.724 -> 654533: EV_JOINED
12:23:03.724 -> netid: 19
12:23:03.724 -> devaddr: <>
12:23:03.724 -> AppSKey: <>
12:23:03.724 -> NwkSKey: <>
12:23:03.724 -> 654759: EV_TXSTART
````
[image]

**Environment**

- LMIC = v.3.1.0
- Arduino IDE 1.8.12
- TTN 
- US915
- Adafruit Feather M0 LORA (includes radio) 

Any thoughts? I'm happy to provide more info if it would be helpful.  

Thank you!

pomplesiegel, 2020-03-03:
Another detail: The same behavior occurs both when joining a "The Things Gateway" gateway or "Multitech Conduit"

terrillmoore, 2020-03-03:
This looks more like #523. It could possibly be the lack of DIO1 to D6 for the "out of the box Feather". To rule out #523, you could try reverting https://github.com/mcci-catena/arduino-lmic/commit/306330fcb5be10c9897f96d2885af530d168a5d3. Also, please list your Adafruit BSP version. I had reports of a recent Adafruit BSP breaking things which I didn't investigate because they switched to the MCCI BSP. 

The root cause of all these problems is the interrupt disable in `os_radio()` (#524). 

With code without 306330f, it didn't hang for me, but I got grossly inaccurate times (as measured externally). As soon as I can get a few free days, I'll correct #524, but I'm extremely busy with another project right now.

pomplesiegel, 2020-03-03:
@terrillmoore, thanks so much for jumping on this. 

I tried reverting just 306330f but there are conflicts in hal/hal.cpp. Please advise what I should include. 

Regarding the BSP, I believe it is 1.5.11 (I had the same behavior with 1.5.10). Screenshot below to confirm this is the right value:
[image]

BTW: Are you recommending an alternative wiring pattern for the Feather (or more generally, a RFM95W to a atsamd21g18) for better performance/compatibility? Perhaps I misunderstood.

terrillmoore, 2020-03-04:
No, just pointing out that the Adafruit Feather M0 is missing a critical connection for the LMIC: DIO1 from the SX1276 is not connected to anything. See the README.md for more info. The LMIC for the Feather M0 assumes that someone has wired DIO1 to D6, but the user has to remember to do that.

pomplesiegel, 2020-03-05:
Got it. Thank you for the clarification. 

OK, sounds good. I will plan to use 2.3.2 until the interrupt issue is addressed. No problem.

KiwiGonzales, 2020-03-22:
> No, just pointing out that the Adafruit Feather M0 is missing a critical connection for the LMIC: DIO1 from the SX1276 is not connected to anything. See the README.md for more info. The LMIC for the Feather M0 assumes that someone has wired DIO1 to D6, but the user has to remember to do that.

I have the same problem with 3.1.0 - but I've connected D01 and D6 with a wire. But even with that it doesnt worked.
With 3.0.99 it still works

trlafleur, 2020-05-06:
Same issue here... works fine with older versions, not with 3.1.0...

pomplesiegel, 2020-05-06:
I just tested and things still work for me using the tagged release
https://github.com/mcci-catena/arduino-lmic/releases/tag/V3.0.99.6

however, things break with the next release
https://github.com/mcci-catena/arduino-lmic/releases/tag/v3.0.99.10

pomplesiegel, 2020-05-13:
@terrillmoore Testing right now, but everything appears to be working swimmingly using v3.2! Thank you!

terrillmoore, 2020-05-13:
@pomplesiegel Excellent, thanks for letting me know. If you think of it, close the issue when you're satisfied that things are fixed.

pomplesiegel, 2020-05-13:
Great! Closing now, as this particular issue seems to be totally eliminated. Thanks again!

closed 2020-05-13 by pomplesiegel

IoTThinks, 2020-05-15:
> @terrillmoore Testing right now, but everything appears to be working swimmingly using v3.2! Thank you!

I also have DIO1 disconnected.
Hence, finally, do you need DIO1 connected to make it sending again?

Thanks a lot.