mcci-catena/arduino-lmic #551: Set TX power

usernamevalis, 2020-03-04:
**Describe your question or issue**
How do I set the TX power of my module?
If someone could point me to the relevant code.

**Environment**

This information is all important; it's hard to help without a complete set of answers.

- Version of LMIC being used: latest
- Version of Arduino IDE being used: 1.8.10
- Network provider Custom network
- Region EU868
- Board custom: SAMR34
- Radio SX1276

terrillmoore, 2020-03-15:
The LMIC sets the transmit power automatically based on information from the network. (ADR is enabled by default.) Since this is a SAMR34, if you're not seeing adequate TX power, it's likely that your module pin-mapping is not optimized. There are sketchy instructions in the [README](https://github.com/mcci-catena/arduino-lmic#advanced-initialization) file's "[advanced initialization](https://github.com/mcci-catena/arduino-lmic#advanced-initialization)" section.

FedericoAndrades, 2020-07-13:
Hello, I am wanting to run the SAMR34 on the AU915 band, and I cannot. The code runs and with the DEBUG active I see that it communicates with the SX1276 but the uplinks never reach the gateway.
I am using a "SAMR34 Xplained Pro" demo board with BASTWAN soft as a base.
If you had a similar project where you can assess what I am doing wrong, I would appreciate it.

sabas1080, 2021-02-11:
@FedericoAndrades for BasTWAN I have added support in our fork working https://github.com/ElectronicCats/arduino-lmic

cnmicha, 2021-06-15:
Related to #23