mcci-catena/arduino-lmic #557: uninitalized variables in os_aes()

cyberman54, 2020-03-15:
There are some variables in `os_aes() `in `lmic.c` that maybe used uninitialized, says my compiler.

t1 in

`t1 = (t1<<8) | ((t0<len) ? buf[t0] : (t0==len) ? 0x80 : 0x00);`

and a0, a1, a2, a3 in
                 
```
if( mode & AES_MIC ) {
                    a0 ^= AESAUX[0];
                    a1 ^= AESAUX[1];
                    a2 ^= AESAUX[2];
                    a3 ^= AESAUX[3];
                }
```

terrillmoore, 2020-03-15:
I see you're using `USE_ORIGINAL_AES`, not the default `USE_IDEETRON_AES`. That's why I've not seen this.

The first looks like a false positives... (Or in the case of t1, there undetermined bits in bits 31..8, but not in bits 7..0,; and this is a loop that repeatedly shift's left. It ultimately has no effect because the undetermined bits are shifted off the top of the word.)

The second: I dunno, sorry, I can't read the code... and I don't use it. I assume (if this is working for you) that it's a false positive, but I'd have to step through the code to figure out what the correctness-preserving transformation is. I suspect that one could move line 274 to 265, but ...

cyberman54, 2020-03-15:
Yes, AES is weird... - i will switch to Ideetron AES.
Maybe a reason to bring up issue #202 again.

terrillmoore, 2020-05-12:
I plan to add proper support for AES from a secure element, and so this might all get resolved. However, if anyone has time to figure out how to clean up the warnings in this code, it would be welcome.