mcci-catena/arduino-lmic #561: os_init(); Failure returned>oslmic.c:53

DerekLF, 2020-04-14:
I am new to LORA, and I'm more of an hardware engineer so programming isn't my strong suit.

After finally managing to upload it without an compile errors I get:
```
Starting
FAILURE 
C:\Users\...\src\lmic\oslmic.c:53
```
After an quick search I found the error points to ASSERT(0);
```
void os_init() {
    if (os_init_ex((const void *)&lmic_pins))
        return;
    ASSERT(0);
}
```
#40 started with an similar issue, and it was all I could find that could be related.

I'm using the latest version of the LMIC library(downloaded earlier this morning)
Arduino IDE 1.8.9
Board is the TTGO LORA32(SX1276): EU868
I'm planning on connecting to an TTN gateway, but I think

My code (slightly modified and trimmed example code):
```
#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>
#define LMIC_PRINTF_TO Serial
#define LMIC_DEBUG_LEVEL 2

static const u1_t PROGMEM APPEUI[8]={  (I'll keep this to myself)  , 0xD5, 0xB3, 0x70};
void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}

static const u1_t PROGMEM DEVEUI[8]={  (I'll keep this to myself)  };
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}

static const u1_t PROGMEM APPKEY[16] = {  (I'll keep this to myself)  };
void os_getDevKey (u1_t* buf) {  memcpy_P(buf, APPKEY, 16);}

static uint8_t mydata[] = "Hello, world!";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 60;

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 18, 
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 23,
    .dio = {26, 33, LMIC_UNUSED_PIN} 
};

void printHex2(unsigned v) {
    v &= 0xff;
    if (v < 16)
        Serial.print('0');
    Serial.print(v, HEX);
}

void onEvent (ev_t ev) {
    Serial.print(os_getTime());
    Serial.print(": ");
    switch(ev) {
        case EV_SCAN_TIMEOUT:
            Serial.println(F("EV_SCAN_TIMEOUT"));
            break;
        case EV_BEACON_FOUND:
            Serial.println(F("EV_BEACON_FOUND"));
            break;
        case EV_BEACON_MISSED:
            Serial.println(F("EV_BEACON_MISSED"));
            break;
        case EV_BEACON_TRACKED:
            Serial.println(F("EV_BEACON_TRACKED"));
            break;
        case EV_JOINING:
            Serial.println(F("EV_JOINING"));
            break;
        case EV_JOINED:
            Serial.println(F("EV_JOINED"));
            {
              u4_t netid = 0;
              devaddr_t devaddr = 0;
              u1_t nwkKey[16];
              u1_t artKey[16];
              LMIC_getSessionKeys(&netid, &devaddr, nwkKey, artKey);
              Serial.print("netid: ");
              Serial.println(netid, DEC);
              Serial.print("devaddr: ");
              Serial.println(devaddr, HEX);
              Serial.print("AppSKey: ");
              for (size_t i=0; i<sizeof(artKey); ++i) {
                if (i != 0)
                  Serial.print("-");
                printHex2(artKey[i]);
              }
              Serial.println("");
              Serial.print("NwkSKey: ");
              for (size_t i=0; i<sizeof(nwkKey); ++i) {
                      if (i != 0)
                              Serial.print("-");
                      printHex2(nwkKey[i]);
              }
              Serial.println();
            }
            // Disable link check validation (automatically enabled
            // during join, but because slow data rates change max TX
	    // size, we don't use it in this example.
            LMIC_setLinkCheckMode(0);
            break;
        case EV_JOIN_FAILED:
            Serial.println(F("EV_JOIN_FAILED"));
            break;
        case EV_REJOIN_FAILED:
            Serial.println(F("EV_REJOIN_FAILED"));
            break;
        case EV_TXCOMPLETE:
            Serial.println(F("EV_TXCOMPLETE (includes waiting for RX windows)"));
            if (LMIC.txrxFlags & TXRX_ACK)
              Serial.println(F("Received ack"));
            if (LMIC.dataLen) {
              Serial.print(F("Received "));
              Serial.print(LMIC.dataLen);
              Serial.println(F(" bytes of payload"));
            }
            // Schedule next transmission
            os_setTimedCallback(&sendjob, os_getTime()+sec2osticks(TX_INTERVAL), do_send);
            break;
        case EV_LOST_TSYNC:
            Serial.println(F("EV_LOST_TSYNC"));
            break;
        case EV_RESET:
            Serial.println(F("EV_RESET"));
            break;
        case EV_RXCOMPLETE:
            // data received in ping slot
            Serial.println(F("EV_RXCOMPLETE"));
            break;
        case EV_LINK_DEAD:
            Serial.println(F("EV_LINK_DEAD"));
            break;
        case EV_LINK_ALIVE:
            Serial.println(F("EV_LINK_ALIVE"));
            break;
        case EV_TXSTART:
            Serial.println(F("EV_TXSTART"));
            break;
        default:
            Serial.print(F("Unknown event: "));
            Serial.println((unsigned) ev);
            break;
    }
}

void do_send(osjob_t* j){
    // Check if there is not a current TX/RX job running
    if (LMIC.opmode & OP_TXRXPEND) {
        Serial.println(F("OP_TXRXPEND, not sending"));
    } else {
        // Prepare upstream data transmission at the next possible time.
        LMIC_setTxData2(1, mydata, sizeof(mydata)-1, 0);
        Serial.println(F("Packet queued"));
    }
    // Next TX is scheduled after TX_COMPLETE event.
}

void setup() {
    Serial.begin(9600);
    Serial.println(F("Starting"));

    // LMIC init
    os_init();
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();

    // Start job (sending automatically starts OTAA too)
    do_send(&sendjob);
}

void loop() {
    os_runloop_once();
}
```

altishchenko, 2020-04-14:
@DerekLF The ASSERT you are hitting means your radio modem is not setup correctly. It is the only way you will get this error. Check the pinmap, make sure it corresponds to your board layout.
radio will fail to initialize if you either have a wrong modem defined in your config or your modem returns the wrong value when asked for its model.

altishchenko, 2020-04-14:
Share your project config please, to see if everything is setup as it should. Which example you modified?

DerekLF, 2020-04-14:
ai forgot to mention, I've modified the ttn-otaa.ino example file.
I used the pinmap mentioned [here](https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-3/18436)
This is the lmic_project_config.h: 
```
// project-specific definitions
#define CFG_eu868 1
//#define CFG_us915 1
//#define CFG_au915 1
//#define CFG_as923 1
//#define LMIC_COUNTRY_CODE LMIC_COUNTRY_CODE_JP	/* for as923-JP */
//#define CFG_kr920 1
//#define CFG_in866 1
#define CFG_sx1276_radio 1
//#define LMIC_USE_INTERRUPTS

```

altishchenko, 2020-04-14:
And did you figure out which board version you have? According to the link there is just a whole lot of them and some of them are not compatible.. Also, I am a bit worried about 'DIO not connected' story, but this should not affect the init part.
Can you add the following bit of code to your library file to check one thing?

oslmic.c: after line 1092 `u1_t v = readReg(RegVersion);` and before #ifdef:

```

   LMIC_DEBUG_PRINTF("Modem version: %02X\n", v);
```

DerekLF, 2020-04-14:
I have the T3 V2.16. The only difference between TTGO LORA32 models seem to be some hardware modifications. I found the schematics of the board, and it seems like both DIO are connected via the board

I might be wrong but my oslmic.c only has 168 lines, and doesn't contain 
> u1_t v = readReg(RegVersion);

or do you mean an different file?

altishchenko, 2020-04-14:
SORRY! I was going to write radio.c file, but got carried away :)

DerekLF, 2020-04-14:
NP, Changed it. but the serial output hasn't changed

altishchenko, 2020-04-14:
You should put LMIC_DEBUG_LEVEL 2 line in your project_config file and recompile everything.

DerekLF, 2020-04-14:
```
Starting
Modem version: 00
FAILURE 
C:\Users\...\src\lmic\oslmic.c:53
```
at least something has changed

altishchenko, 2020-04-14:
Yep. This means that there is something wrong with pins.
Modem is not replying anything. Now, going one level higher (the sketch itself is ok and pinmap in it too), _how_ did you make Arduino compile code for your board? The reason I am asking is that using the wrong Board/Model during compile may result in the wrong mappings of the MISO/MOSI/SCK pins for the SPI to work properly.

altishchenko, 2020-04-14:
Here is an ESP32-Arduino link, in case you don't have it already [ESP32 support in Arduino IDE](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md)

DerekLF, 2020-04-14:
Okey that's odd, first the sketch wouldn't compile on board type: TTGO LORA32-oled V1 
so I chose ESP32 Dev Module, but now it does and it dumps:
```
Starting
Modem version: 12
RXMODE_RSSI
4059: engineUpdate, opmode=0x8
Packet queued
4072: EV_JOINING
4076: engineUpdate, opmode=0xc
7163: engineUpdate, opmode=0xc
7180: EV_TXSTART
7247: TXMODE, freq=868100000, len=23, SF=7, BW=125, CR=4/5, IH=0
322862: setupRx1 txrxFlags 00 --> 01
start single rx: now-rxtime: 4
323494: RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 324022 rxtime: 323487 entry-rxtime: 535 now-entry: 4 rxtime-txend: 312375
385362: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 4
385994: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 400346 rxtime: 385987 entry-rxtime: 14359 now-entry: 3 rxtime-txend: 374875
400363: processRx2Jacc txrxFlags 0x2 --> 00
401103: Unknown event: 20
402861: engineUpdate, opmode=0xc
```

altishchenko, 2020-04-14:
Cool! You may remove LMIC_DEBUG_LEVEL from your code, recompile and see if everything works as needed!

altishchenko, 2020-04-14:
V1 in your comment worries me a bit. You have V3 board and that is a lot in electronics :)

DerekLF, 2020-04-14:
[image]
aside from some relocated parts and swapped GPIOs not much has changed between versions.

```
Starting
Packet queued
4147: EV_JOINING
251956: EV_TXSTART
645134: Unknown event: 20
4584183: EV_TXSTART
4977356: Unknown event: 20
8698853: EV_TXSTART
9092025: Unknown event: 20
```
is Unkown event: 20 related to not being able to connect with the gateway? (I think I'm only a bit out of range, the gateway is located 15km away)

altishchenko, 2020-04-14:
Nope. :^) It says that the board has sent Join request out over the radio. Just that. Event 20 is EV_JOIN_TXCOMPLETE.
You can try one more "tune" that may save you some time: in the file lmic_bandplan_eu868.h, line 64, change SF7 to SF12, so it will read:
```
#define LMICbandplan_getInitialDrJoin() (EU868_DR_SF12)
```

altishchenko, 2020-04-14:
SF12 has longer time on air, but also better suited to long distances. LMIC library tries to start 'high' with better speeds/shorter distances and then gradually falls down to lower speeds. Setting SF12 will just save time and it won't affect real data transmission, just network connection.

DerekLF, 2020-04-14:
I'll leave it running for an while. But I think I'll have to test it when I'm closer to the gateway. So far it hasn't picked up any data. The range of the SX1276 only seems to be around the 6km

altishchenko, 2020-04-14:
Yes, 15 is a little too far :)

DerekLF, 2020-04-14:
It was supposed to be only 100 meters, just on the other side of the school grounds. But yeah, I cant really go to school. Even though this is part of my exam. Thanks for all the help!

altishchenko, 2020-04-14:
Happy testing and good luck with exam! If it works for you, please don't forget to close the issue.

terrillmoore, 2020-05-12:
Is this still live? If not, may I close it?

altishchenko, 2020-05-12:
Yeah, it's closed.

sxwid, 2020-11-18:
As this issue is still open - I have the same. I try to run the ttn-abp demo code on a Arduino UNO with a Dragino LoRa Shield v1.4. os_init(); is failing with the exact same error. In my understanding it is a problem of wiring. But according to the official wiring [here](http://wiki.dragino.com/index.php?title=File:LoRa_Shield_Pin_Mapping.png) the codesnippet should be ok:
```
const lmic_pinmap lmic_pins = {
  .nss = 10,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = 9,
  .dio = {2, 6, 7},
};
```
Debug Mode Level 2 is not possible due to not enogh memory. Level1 doesn't show interesting stuff. Or I am looking at the wrong place. Is there another approach I can take?

closed 2020-12-10 by terrillmoore