mcci-catena/arduino-lmic #562: Travis CI fails pull request checks because of the error in one of the examples

altishchenko, 2020-04-15:
Hello! I opened this issue as a support request because other choices are no better. There is a problem with the current configured Travis-CI regression tests and one of the examples in the library, which prevents successful checks of the filed PRs. As far as I could dig out from the Travis logs, here is a snapshot, that may help tracking it down:
Building in release mode
Compiling .pio/build/heltec_wifi_lora_32/src/ttn-otaa-feather-us915-dht22.ino.cpp.o
Generating partitions .pio/build/heltec_wifi_lora_32/partitions.bin
Compiling .pio/build/heltec_wifi_lora_32/lib494/Adafruit ADXL343_ID6117/Adafruit_ADXL343.cpp.o
.pio/libdeps/heltec_wifi_lora_32/Adafruit ADXL343_ID6117/Adafruit_ADXL343.cpp:27:18: fatal error: Wire.h: No such file or directory
**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************
 #include <Wire.h>
                  ^
compilation terminated.
*** [.pio/build/heltec_wifi_lora_32/lib494/Adafruit ADXL343_ID6117/Adafruit_ADXL343.cpp.o] Error 1
========================== [FAILED] Took 6.19 seconds ==========================
The command "ci/platformio.sh" exited with 1.

terrillmoore, 2020-05-10:
Sorry, I don't use PlatformIO myself. Perhaps someone using PlatformIO can investigate and help out. Obviously, this is due to an external change that must be tracked. (This is one of the reasons I don't use PlatformIO -- it doubles the amount of work for tracking external changes.) I will check the logs and find the contributor and perhaps that person can assist.

terrillmoore, 2020-05-10:
@frazar, any chance you could look at this? I know this is a busy time for a lot of people...

PR #569 (2020-05-10): Update library dependencies for some examples
This PR adds the "Wire" library to the list of required libraries for building examples that rely on the "DHT sensor library". The list of required libraries is specified with the `lib_deps` option for PlatformIO.

Additionally, the "Adafruit Unified Sensor" library is removed from the same list. This is because PlatformIO automatically recognizes it as a required dependency of the "DHT sensor library".

Fixes #562
closed 2020-05-10 by terrillmoore