mcci-catena/arduino-lmic #566: timing issue and crashes during rx1 window data processing on STM32F103

dolfandringa, 2020-04-30:
**Describe the issue**

I am using an STM32F103 based board (maple mini clone) using Chirpstack and [Congduc Pham's LowCostLoraGw|https://github.com/CongducPham/LowCostLoRaGw] as packet forwarder. I am using the abp example but only with a single channel (yes i know single channel gateways are not lorawan compliant but don't have any other option for now). For some reason, I am getting crashes during the RX1 window, as soon as it is trying to process the RX1 data. It kind of feels like a timing issue again like issue #563, but in this case, adding a delay or debug print statements doesn't fix it fully.  With no or a delay<10ms at the end of the `loop()` function, EV_TXCOMPLETE never happens. But with a delay >=10ms, after about 4-7 frames, the microcontroller freezes as well, between EV_TXSTART and EV_TXCOMPLETE.

**Environment**

- Version of LMIC being used: 3.1.0
- Version of Arduino IDE being used: Platformio 4.2.1 with arduino 1.8 and Roger Clark Melbourne stm32duino maple core
- Network provider: Local chirpstack with LowCostLoraGw packet forwarder
- Region: US915
- Board (MCCI Catena, Adafruit Feather M0, Heltec, etc.):  maple mini
- CPU : STM32F103C8
- Radio (HopeRF, SX1276, etc.): RFM96W

terrillmoore, 2020-04-30:
My suggestion is that you get an STLINK V2, and use GDB to debug the crash location. There are instructions at https://github.com/mcci-catena/Catena-Sketches/blob/master/extra/HOWTO-DEBUG-WITH-GDB.md -- these are not actually specific to the Catena board. With that, you can single-step, set breakpoints, etc. Cost for a "genuine STLINK V2" is about $22. Knock-offs are even cheaper. DigiKey has them, and I suppose Amazon has them too.

altishchenko, 2020-05-01:
@dolfandringa I'd also recommend Keil as a debugger, it is free for STM32x0 boards, but yours is F1, no luck. I stumbled once upon the issue of sudden reboots  where func() pointer in os_runloop_once() became NULL after interrupts enable call and solved it like this in my oslmic.c version (look for job->ready):
```
// unlink job from queue, return if removed
static int unlinkjob (osjob_t** pnext, osjob_t* job) {
    for( ; *pnext; pnext = &((*pnext)->next)) {
        if(*pnext == job) { // unlink
            *pnext = job->next;
            job->ready = false; // <-- invalidate
            return 1;
        }
    }
    return 0;
}
...
void os_runloop_once() {
    osjob_t* j = NULL;
    hal_disableIRQs();
    // check for runnable jobs
    if(OS.runnablejobs) {
        j = OS.runnablejobs;
        OS.runnablejobs = j->next;
        j->ready = true; // <-- Put mark
    } else if(OS.scheduledjobs && hal_checkTimer(OS.scheduledjobs->deadline)) { // check for expired timed jobs
        j = OS.scheduledjobs;
        OS.scheduledjobs = j->next;
        j->ready = true; // <-- Put mark
    } else { // nothing pending
        hal_unlock(); // Allow CPU going low power
    }
    hal_enableIRQs();
    if(j && j->ready) { // <-- check the function was not invalidated during hal_enableIRQs()
        j->func(j);
    }
}
```

dolfandringa, 2020-05-04:
hey guys, awesome thanks for the help. I have an STLink coming in already.7 No idea how fast delivery will go. The NULL pointer seems like a likely cause. I will check it out. In the mean time I did get a RAKKK2245 concentrator as well, so will have a proper gateway. Will comment back here.

dolfandringa, 2020-05-05:
It is interesting, switching to a full rak2245 concetrnator solved a lot of it it seems. Still need the delay(2), but at least I managed to continue sending 80 messages in a row without interruption, so part of it seems to be something with the combination of the gateway and LMIC. I will close it for now and keep monitoring it.

closed 2020-05-05 by dolfandringa