mcci-catena/arduino-lmic #567: Will you consider to add sx1262 support?

RAKWireless, 2020-05-08:
Hi, 
We are now designing a new product, which contains nRF52840 and sx1262 for lorawan. Arduino-lmic works well on our products with sx1276. We think this is the most integrated arduino lorawan lib. So do you have any plan to add sx1262 support to this lib? it will be great. Thank you!!

terrillmoore, 2020-05-08:
Hi, thanks for your note and your kind words.

We have development boards for the sx1262, and we have studied the work required. It looks straightforward. Unfortunately, because of the virus situation, we are very resource constrained. Hope to get to this in the next month or two.

Best regards,
--Terry

RAKWireless, 2020-05-14:
Hi @terrillmoore ,
Can you please share some idea on how to support SX1262?
We want to have a try based on your repo and your guidance, then pull request in this repo if we success. :)
Thank you!

terrillmoore, 2020-05-18:
Hi @RAKWireless -- thanks for your interest.

I've been thinking about this on the architecture side, but I'd welcome outside help on the implementation side, if you're willing to put up with my quirks....

There are three preliminary implementation steps that have to be done.

1. header file for the SX126x registers and commands. This is just typing; but we should be careful to follow the names in the Semtech data sheet, to assist with searching and cross referencing. We'll need tentative file names.  It would be really good if we checked the SX128x at the same time -- I am very interested in 2.4GHz support, and I hope we can get close.

2.  Check https://github.com/mcci-catena/arduino-lmic/blob/master/doc/RadioDriver.md, which defines the current APIs for radio.c.  (There are probably some errors, because I don't think anyone but me has ever read the document.) Look for gaps between SX127x and SX1262 function. I think, generally, that the SX126x driver is a closer fit to the APIs than the SX127x is.  

3. Figure out how whether the SX126x needs any additional state variables in `struct lmic_t` -- these would ideally go in a union type in `lmic_radio_data_t`.  We'd refer to them (typing a lot, but very portably) as `LMIC.radio.sx126x.{fieldname}` -- and we'll move any SX127x state into `LMIC.radio.sx127x.{fieldname}` when we put things together.

My architectural goal is to minimize `#if` directives in the C files, but still make things compile down small (the way they do now).

Let me know if this is still of interest, and makes sense. We may need to set up an email reflector to handle implementation discussions, because the github issue workflow doesn't mesh very well with my daily workflow while doing $DAYJOB, and I don't want to be too much of a bottleneck. Chat-based workflows are also not very good for me, because my attention has to be focused on other things, and then I miss it... If you're still interested, I'll think about how best to facilitate this.

Best regards,
--Terry

lnlp, 2020-10-04:
Hi @terrillmoore,
I noticed no responses since your last message in this thread.
Has there been any progress on SX1262 support since May?

Thanks

terrillmoore, 2020-12-10:
> Has there been any progress on SX1262 support since May?

None...

closed 2020-12-10 by terrillmoore