mcci-catena/arduino-lmic #579: Dragino Lora shield does not support floating RST

cc32d9, 2020-05-23:
Board description: https://wiki.dragino.com/index.php?title=Lora_Shield

Issue in detail: https://github.com/manuelbl/ttn-esp32/issues/23

Summary: the board does not have a pull-up resistor on RST, so the pin needs to be always driven in order for Lora chip to function. So, a new configuration option is needed that sets the RST pin to always-on instead of leaving it floating.

I proposed a workaround for ttn-esp32 library, but I believe it will be much better to have the fix in arduino-lmic.

terrillmoore, 2020-05-23:
Sorry you're having problems.

Unfortunately, please see the SX1276 data sheet. It explicitly says "leave RESET floating" (section 7.2). So enabling this would cause people to accidentally violate SX1276 datasheet. That's a bad idea. It doesn't matter if it might work in practice; we can't ship code that violates the hardware datasheet.

There is obviously a *pull-down* resistor on your board; you should look at the schematic. It's possibly that your BSP is enabling a built-in pull-down (inside the ESP32). In any case, you need to fix your hardware so we can operate the SX1276 properly.

[image]
[image]

The SX1272 has similar requirements, except that it's inverted.
[image]
[image]

cc32d9, 2020-05-24:
thanks, I'll address that to the hardware maker

terrillmoore, 2020-05-24:
Sorry I was so abrupt. We can tag this as an enhancement request. I still think changing the release is a bad idea, but I have no problem suggesting a patch you can make to fix your setup.

cc32d9, 2020-05-24:
well I don't know what's the best solution for this. There are thousands of such boards already manufactured. Hopefully Dragino fixes the issue in the next version. And there's not much space on the board to add a pull-up resistor manually. 

I would think a configuration option, like `CFG_sx127x_rst_pullup` would be good. It won't change the existing default behavior, but would allow a workaround for boards like this.

cc32d9, 2020-05-24:
The hardware manufacturer informed me that they will fix this in the next release. But there are thousands of such boards on the market with this bug. The patch that I proposed is only suitable for ESP32 chips, but this shield is also widely used with other Arduino boards.

So, I think there needs to be a configurable option that leaves RST driven up after the chip initialization.

terrillmoore, 2020-05-24:
Hi, really sorry but I'm not willing to take on the (forever) support burden of incorporating this workaround into the code base. My resources are very limited and there are important things that need to be done (SX1262 support, Class B&C, etc.) No matter how many boards are out there, LoRaWAN boards are much less expensive than engineering time. Thanks for your understanding.

terrillmoore, 2020-05-24:
By the way, you can work around this *in your sketch*, I think, with out any source changes. 

1. configure NRST as a *no connect*. 
2. manually reset the radio before initializing the LMIC.

Thanks,
--Terry

cc32d9, 2020-05-24:
The hardware vendor informed that they use https://github.com/dragino/arduino-lmic  for testing, which is quite similar and it's also leaving the RST floating.

So, it could either be a faulty piece of hardware, or the combination with Wemos board that caused this issue.  Two more such Lora shields are on the way to me, so I will be able to say if it's persistent.

closed 2020-12-10 by terrillmoore