mcci-catena/arduino-lmic #585: ABP mode gives an ASSERT on LMIC_setTxData2.

DaveDavenport, 2020-06-23:
**Describe your question or issue**

Because of problems with #311 I was trying out ABP mode.  I am getting the following error on do_send (from the examples).

```
~/Arduino/libraries/arduino-lmic-master/src/lmic/radio.c:1065
```
[here](https://github.com/mcci-catena/arduino-lmic/blob/master/src/lmic/radio.c#L1065)

( ttn-abp )

I actually see the message arrive in  the dashboard (I updated the do_send to send out some specific data):

[image]

**Environment**

* TRFM95W module.. uses SX1276 I think connected to a blue pill (stm32f108).
* On EU network. 868 MHz. TheThingsNetwork
* GIT version of 2020-06-23.

On a ttgo t3 it does seem to work. 

So I am looking at hints to see where the actual problem lies?

terrillmoore, 2020-06-23:
The radio should automatically drop to sleep mode after a successful tx. It's not. So ... looks like a hardware problem to me.

DaveDavenport, 2020-06-23:
thx, was afraid of that.  I am going to try another module.

closed 2020-06-23 by DaveDavenport