mcci-catena/arduino-lmic #587: Saving / Loading state

runezor, 2020-07-02:
I'm currently using this library on ESP devices, and when using deep sleep on these devices, the memory is cleared. It's often a hassle having to restore the entire state of this library, and I find myself writing structs in my code that just mirror the LMIC object.

Instead I think it'd be nice if the library supported this natively.

I'm currently working on mirrorring the state in an arduino sketch, but if the author feels like this is a worthwhile addition to the library, I wouldn't mind making a pull request. I can think of two approaches:
Creating a method in the LMIC object that just returns its current state as a struct, along with a method to initialise the object using the struct.
Creating a method in the LMIC object that can save and load its current state using callback functions (to a function that does something like saving/loading a single block).

terrillmoore, 2020-07-03:
Thank you for opening this discussion.

I would have done this long ago, but the LMIC implementation makes all of the contents of the LMIC structure part of the API. So we can't do this, alas, without a careful procedure (and some number of months while people prepare and adapt).

The first part of the long-term procedure is to write getters for all API LMIC elements, and setters where necessary.  These can be static.

The second part is to rename *everything* (so that this won't have to happen again), and *hide* the LMIC structure contents from the clients. That step could do the reorganization.,

The modest, not so disruptive step, along the lines of what you are discussing is to add state-save and state-restore, and design these to externalize the data (at least far enough that different versions of the LMIC can deal with a common form of saved state).

Despite the need for code space efficiency, let's avoid adding structs that external clients can read; let's keep the saved state opaque to clients. (Remember that people on at32u4s are not using this now, and are surviving; putting all the mechanism in separate functions will ensure they are not penalized.)  If clients need to know what's in a saved state blob, we can provide additional methods.

runezor, 2020-07-06:
Thank you for your response.

While I do agree that following the procedure you describe would result in a much nicer code base, is there anything preventing someone from writing a simple save/restore method without hiding the struct from clients? A bit like the savestate struct in this sketch https://github.com/Edzelf/LoRa/blob/master/ESP_lora_tracker/ESP8266_loratracker.ino#L64, but internal in the library rather than external.

cnmicha, 2021-01-07:
See https://github.com/manuelbl/ttn-esp32/issues/30#issuecomment-683439753

terrillmoore, 2021-06-01:
Rather than add the code in C, I've done this in C++ in the [Arduino LoRaWAN library](https://github.com/mcci-catena/arduino-lorawan). Perhaps that would be enough?

lnlp, 2021-08-14:
> @terrillmoore terrillmoore added this to To do in v4.0 on Feb 22

It's currently August 14, 2021 and arduino-lmic v4.0 was released some while ago but currently still no support for saving/restoring LMIC state.
<br>

> Rather than add the code in C, I've done this in C++ in the Arduino LoRaWAN library. Perhaps that would be enough?

No, not at all.

We desparately need support for saving/storing LMIC state _**in**_ the LMIC library.
See: https://forum.mcci.io/t/lmic-development-plans/19/6

> Rather than add the code in C, I've done this in C++

Whether implemented in C or C++ is less of a concern, more important is to have the functionality available in the LMIC library.
_Actually I prefer C++ because a higher level language with its advantages._

LMIC users should be able to use standard generic functions (e.g. `saveLmicState(...)` and `restoreLmicState(...)`) that are part of the LMIC library and then implement their own code to physically store and restore it to/from some kind of non-volatile memory. Actual implementation will depend MCU architecture and additional hardware (e.g. availability of EEPROM, flash or FRAM) and therefore needs to be implemented by the user, but internal support to enable this needs to be provided by the LMIC library itself.

> I've done this in C++ in the [Arduino LoRaWAN library](https://github.com/mcci-catena/arduino-lorawan). Perhaps that would be enough?

No, support has to be provided (and documented) in the LMIC library itself.

Users should only have to implement the (standard, generic) `saveLmicState()` and `restoreLmicState()` functions. Users should not have to be bothered with having to dive into and trying to understand the arduino-lorawan library first and extract/adapt the parts responsible for saving and restoring LMIC state.

My assumption is that for the author of both MCCI LMIC ('arduino-lmic') and 'arduino-lorawan' it should be relatively straightforward to copy and adapt functionality for saving/restoring state from 'arduino-lorawan' to 'arduino-lmic'.
For an outsider this will be much more difficult and error prone.
It's much better and preferred to have this implemented in the LMIC library in a standard way (where this only needs to be done once and benefits all users).

I am often asked for advice about which Arduino LoRaWAN library to use because users need to save and restore LMIC state but MCCI LMIC does currently not support it. We all would welcome to have this support in the MCCI LMIC library.