mcci-catena/arduino-lmic #589: Is reset pin required for lmic_pins?

KnightOfNih, 2020-07-15:
Hello all,

My challenge feels almost exactly similar to the following issue:

https://github.com/mcci-catena/arduino-lmic/issues/561

```
Modem version: 00
FAILURE 
.pio/libdeps/esp32dev/MCCI LoRaWAN LMIC library/src/lmic/oslmic.c:53
```

My pin setup is as follows:

```
#define SCK 14
#define MISO 12
#define MOSI 13
#define NSS 16
#define DIO0 26
#define DIO1 33
#define DI02 32
#define RST LMIC_UNUSED_PIN

const lmic_pinmap lmic_pins = {
  .nss = NSS,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = RST,
  .dio = {DIO0, DIO1, DI02},
};
```

Is the reset pin required?

On line 1104 in radio.c

`hal_pin_rst(2); // configure RST pin floating!

But my reset pin is unused.

**Environment**

This information is all important; it's hard to help without a complete set of answers.

- Version of LMIC being used:  3.2.0
- Version of Arduino IDE being used: Platform IO
- Network provider: None yet
- Region (EU868, US915, etc.): US915
- Board: Custom board using Espressif ESP32-WROOM-32D
- Radio: RFM95HW

platform.io file

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200

lib_deps =
    MCCI LoRaWAN LMIC library

build_flags =
    -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS
    -D CFG_us915=1
    -D CFG_sx1276_radio=1
```

terrillmoore, 2020-07-15:
The reset pin is required for this library. You're not getting a proper initial state of the radio. Sorry for the inconvenience.

KnightOfNih, 2020-07-15:
Thank you so much for your quick response!  That helps a great deal.  Thank you.

closed 2020-07-15 by KnightOfNih