mcci-catena/arduino-lmic #591: ABP example EV_TXCOMPLETE never gets called so only one packet gets sent

Vilsol, 2020-07-20:
**Describe your question or issue**
I am a beginner to hardware radios so I might be missing something obvious.

I have setup a single channel gateway using the things4u repository: https://github.com/things4u/ESP-1ch-Gateway

I copied the ABP example and only updated the pins to match the board.

The first message gets sent, the gateway receives (but not always) it and it shows up on TTN Console.
After that, there is dead silence. Looking through the example, and following the serial log, it seems that `EV_TXCOMPLETE ` never gets called, therefore it can never be rescheduled.

I am not sure if this is an issue on the device or gateway end. Have I misconfigured the pins on the device? Or is this library incompatible with single channel gateways? Or is that gateway just outdated?

I was trying to find the most up to date information, but most of what I found was >2 years old.

**Environment**

- Version of LMIC being used: 3.2.0
- Version of Arduino IDE being used: PlatformIO Core 4.3.4, Home 3.2.3
- Network provider (The Things Network, Swisscom, ChirpStack, etc.): TTN
- Region (EU868, US915, etc.): EU868
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.): LILYGO TTGO LORA32 V2.0
- Radio (HopeRF, SX1276, etc.): SX1276

terrillmoore, 2020-07-20:
Missing EV_TXCOMPLETE means that the SX1276 chip and the software aren't hooked up correctly -- either the pinmap doesn't match your hardware, or  your hardware is broken.

After that is fixed, you'll find that the single channel gateway has limitations -- for example, you can't disable the mandatory 3 channels in the LMIC, so you're going to miss data 2 out of 3 times because the LMIC doesn't allow the mandatory channels to be disabled. But EV_TXCOMPLETE shoudl always happen, even if nobody's listening.

Good luck!

Vilsol, 2020-07-20:
Thanks for the quick response!

This is the pinout for the board that I am using https://github.com/LilyGO/TTGO-LORA32

From that I came up with this list of pins:
```
const lmic_pinmap lmic_pins = {
    .nss = 18,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 12,
    .dio = {26, 33, LMIC_UNUSED_PIN},
};
```

I took the pin 33 from here https://github.com/mcci-catena/arduino-lmic/blob/master/src/hal/getpinmap_ttgo_lora32_v1.cpp#L22 but now I am starting to think that was a mistake, however I do not see what other pin I could use as DIO1?

Also, maybe you know if there is a way I could use that board in a non-single-channel mode?

terrillmoore, 2020-07-21:
The pin is label IO33 in the schematic, but I have no idea whether digitalRead(33) will return that pin. That can only be determined by looking at the BSP for that device. If in doubt, you can jumper LoRa1 to one of the other pins that you know how to access with digitalRead().  Or you can write a sketch that doesn't use LoRa, and do pinMode(33, OUTPUT) and drive a square wave -- it should cause LoRa1 to wiggle.

The board will work fine with a multi channel gateway, once it works at all. The limitations are in the gateway, not in the device.

Good luck!

closed 2020-12-10 by terrillmoore