mcci-catena/arduino-lmic #593: SAMR34 XplainedPro No Transmit

FedericoAndrades, 2020-07-23:
Sorry for the level of English, I have been helped by Google Translate, sorry.

Using the TTN-OTAA demo, the program does not send messages to the gateway, according to DEBUG (LMIC_DEBUG_LEVEL = 1), the frequencies and DataRate are correct, but no message arrives at the gateway. Nor does it do so in ABP mode.
I have verified the interface I / O and it is correct, verified the SPI communication and it also works, I do not find a problem with the hard (verified with DEMO in ATMEL Studio).
Do you have a project on SAMR34 that I can lean on to investigate my problems.

Thank you.

Perdon por el nivel de ingles, me eh ayudado con el Traductor de Google, perdon.

Usando el demo TTN-OTAA el programa no envia mensajes al gateway, segun el DEBUG ( LMIC_DEBUG_LEVEL = 1 ), las frecuencias y DataRate estan correctas, pero no llega ningun mensaje al gateway. Tampoco lo hace en modo ABP .
Eh verificado el I/O de interfase y esta correcto, verificado la comuniciacion SPI y tambien funciona, no le encuentro problema al hard (verificado con DEMO en ATMEL Studio) .
Tienen algun proyecto sobre SAMR34 en el cual pueda apoyarme para investigar mi problemas.

Gracias.

**Environment**
- Version of LMIC being used: MCCI LoRaWAN LMIC 3.2.0
- Version of Arduino IDE being used: 1.8.13
- Network provider YEAP!
- Region AU915
- Board SAMR34 XplaindPro with BASTWAN ArduinoPackage
- Radio SX1276

cstratton, 2020-08-02:
- Are you able to monitor the gateway for raw packets received, or are you only able to see if your LoRaWAN provided has decoded something they consider to be a valid packet?   What tests have you run with other nodes which indicate the gateway is functioning correctly?

- What have you done to verify that SPI is working correctly?  You might try using a cheap logic analyzer with sigrok

- Consider temporarily modifying the code to print out the raw bytes transmitted and verify in the online lorawan packet decoder that these are valid; this may cause receive timing problems, but uplink has to work before downlink is a concern.

FedericoAndrades, 2020-08-03:
Thanks you for answer, the gateway don´t have monitor for raw packets only LoRaWAN mensage, but i check functionality whit ATMELStudio Demon and work .
To check the SPI, modify the HAL of the SPI to make a reading after each registry write, verifying that the write is done correctly (skip FIFO reg) only as a test, this code does not work during the demo.
I check del lorawan packet send to SX1276 and look ok. 
I also tried the Beelan library (https://github.com/BeelanMX/Beelan-LoRaWAN) and send packages to the gateway without problem, but this library is very limited in functionality and does not help me for my development.

terrillmoore, 2020-08-03:
I'm guessing it's due to the antenna control configuration. That's usually the problem in this kind of issue. (There are a couple of GPIOs for TX/RX enable; there might also be control needed for an external power amplifier or for the TCXO, if any.) It can be tricky without careful examination of reference code.

FedericoAndrades, 2020-08-06:
Hello, the control of the RX / TX is directed by the SX1276 not by the GPIO, on the other hand this circuit does not have an amplifier and the TXCO is working.
Here I attach the schematic of the board that I am using, and the program that I am testing.

> > /*******************************************************************************
>  * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
>  *
>  * Permission is hereby granted, free of charge, to anyone
>  * obtaining a copy of this document and accompanying files,
>  * to do whatever they want with them without any restriction,
>  * including, but not limited to, copying, modification and redistribution.
>  * NO WARRANTY OF ANY KIND IS PROVIDED.
>  *
>  * This example sends a valid LoRaWAN packet with payload "Hello,
>  * world!", using frequency and encryption settings matching those of
>  * the The Things Network.
>  *
>  * This uses OTAA (Over-the-air activation), where where a DevEUI and
>  * application key is configured, which are used in an over-the-air
>  * activation procedure where a DevAddr and session keys are
>  * assigned/generated for use with all further communication.
>  *
>  * Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in
>  * g1, 0.1% in g2), but not the TTN fair usage policy (which is probably
>  * violated by this sketch when left running for longer)!
> 
>  * To use this sketch, first register your application and device with
>  * the things network, to set or generate an AppEUI, DevEUI and AppKey.
>  * Multiple devices can use the same AppEUI, but each device has its own
>  * DevEUI and AppKey.
>  *
>  * Do not forget to define the radio type correctly in config.h.
>  *
>  *******************************************************************************/
> 
> #include <lmic.h>
> #include <hal/hal.h>
> #include <SPI.h>
> 
> // This EUI must be in little-endian format, so least-significant-byte
> // first. When copying an EUI from ttnctl output, this means to reverse
> // the bytes. For TTN issued EUIs the last bytes should be 0xD5, 0xB3,
> // 0x70.
> static const u1_t PROGMEM APPEUI[8]={ 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
> void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}
> 
> // This should also be in little endian format, see above.
> static const u1_t PROGMEM DEVEUI[8]={ 0x20, 0x55, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
> void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}
> 
> // This key should be in big endian format (or, since it is not really a
> // number but a block of memory, endianness does not really apply). In
> // practice, a key taken from ttnctl can be copied as-is.
> // The key shown here is the semtech default key.
> static const u1_t PROGMEM APPKEY[16] = { 0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 ,0x01 };
> void os_getDevKey (u1_t* buf) {  memcpy_P(buf, APPKEY, 16);}
> 
> static uint8_t mydata[] = "Hello, world!";
> static osjob_t sendjob;
> 
> // Schedule TX every this many seconds (might become longer due to duty
> // cycle limitations).
> const unsigned TX_INTERVAL = 60;
> 
> // Pin mapping
> const lmic_pinmap lmic_pins = {
>     .nss = SS,
>     .rxtx = 0,
>     .rst = RFM_RST,
>     .dio = {RFM_DIO0, RFM_DIO1, RFM_DIO2},
> };
[... 121 lines cut ...]
>     // First, disable channels 0-72 (is there 72 ??)
>     for (int channel=0; channel<72; ++channel) {
>        LMIC_disableChannel(channel);
>     }
>     // This means only channels 0-8 are up
>     for (int channel=0; channel<8; ++channel) {
>       LMIC_enableChannel(channel);
>     }
> 
>     // Disable link check validation
>     LMIC_setLinkCheckMode(0);
> 
>     // Set data rate and transmit power (note: txpow seems to be ignored by the library)
>     LMIC_setDrTxpow(DR_SF7,14);
> 
>     // Start job (sending automatically starts OTAA too)
>     do_send(&sendjob);
> }
> 
> void loop() {
>     delay(10);
>     os_runloop_once();
> }

[Schematic_SAMR34_Xplained_Pro.PDF](https://github.com/mcci-catena/arduino-lmic/files/5034931/Schematic_SAMR34_Xplained_Pro.PDF)

closed 2020-12-10 by terrillmoore