mcci-catena/arduino-lmic #597: Low memory warning - ways to optimize?

KonstantinMastak, 2020-08-04:
**Describe your question or issue**

I want to move from matthijskooijman/arduino-lmic to mcci-catena/arduino-lmic for my atmega328p-based project due to LoRaWAN specs compatibility issues. 

However, when I try to compile default ttn-otaa example from MCCI-Catena (no changes except "FILLMEIN" contants), I get the following warning:

====
Sketch uses 23884 bytes (74%) of program storage space. Maximum is 32256 bytes.
Global variables use 1877 bytes (91%) of dynamic memory, leaving 171 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
====

When I compile basically the same ttn-otaa example from matthijskooijman, I get the following:

===
Sketch uses 19986 bytes (61%) of program storage space. Maximum is 32256 bytes.
Global variables use 930 bytes (45%) of dynamic memory, leaving 1118 bytes for local variables. Maximum is 2048 bytes.
===

I have no questions about increased program storage space - this is totally justified. However, increase in global variables usage is dramatic and makes arduino-lmic from mcci-catena unusable on atmega328p in any practical task.

So my question is: is there anything I can do with this doubling of dynamic memory usage like maybe turning off some unnecessary defines, making some small modifications to source code that can move some constants to PROGMEM, etc?

Can you point me to which exact parts of mcci-catena implementation consume twice more dynamic memory in comparison to matthijskooijman implementation?

**Environment**

This information is all important; it's hard to help without a complete set of answers.

- Version of LMIC being used:  current master branch (44778c1)
- Version of Arduino IDE being used: 1.8.1
- Network provider (The Things Network, Swisscom, ChirpStack, etc.): ThingPark (Actility)
- Region (EU868, US915, etc.): RU864
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.): self-made based on Atmega328p
- Radio (HopeRF, SX1276, etc.): SX1276

terrillmoore, 2020-08-04:
Sorry Konstantin.

It's using more memory primarily because buffer sizes for messages increased to support full-size messages. This was required for compliance. If you chase down the max message sizes, I think you'll find how to shrink those back down. In addition, I suggest you turn off PING and BEACON support; those are only for class B, and class B is not currently working.

KonstantinMastak, 2020-08-05:
Terrill, thank you very much for your suggestion!

Turning on DISABLE_PING, DISABLE_BEACONS and turning off LMIC_ENABLE_long_messages resulted in:

Sketch uses 21536 bytes (66%) of program storage space. Maximum is 32256 bytes.
Global variables use 1491 bytes (72%) of dynamic memory, leaving 557 bytes for local variables. Maximum is 2048 bytes.

Which is pretty much enough for my needs.

KonstantinMastak, 2020-08-08:
The problem is solved, thank you.

closed 2020-08-08 by KonstantinMastak