mcci-catena/arduino-lmic #602: Multi region support?

AndreiHanu, 2020-09-11:
**Describe your question or issue**

Hello!

We are currently developing a GPS tracker based on the Murata CMWX1ZZABZ-091 chip and need to be able to select the LoRa frequency at runtime based on the GPS location and a preset geofence. 

Currently, this LMIC defines the region during compile time, but is it possible to change it during runtime? See example code below which has good intentions, but does not work.

`
/ If available, set the LoRa region based on GPS lat and long
  // Otherwise, default to the US region (915 MHz)
  if (tlm_pkt.ValidGPSFix)
  {
    // Set LoRa region based on GPS
//    set_LoRa_Region(55.751244, 37.618423);                  // Moscow, Russia  
//    set_LoRa_Region(51.507351, -0.127758);                  // London, England  
//    set_LoRa_Region(39.904202, 116.407394);                 // Beijing, China 
//    set_LoRa_Region(-1.292066, 36.821945);                  // Nairobi, Tanzania
    set_LoRa_Region(tlm_pkt.Latitude, tlm_pkt.Longitude);
  } 
  else 
  {
    // Default: US frequency
//    set_LoRa_Region(55.751244, 37.618423);                  // Moscow, Russia  
//    set_LoRa_Region(51.507351, -0.127758);                  // London, England  
//    set_LoRa_Region(39.904202, 116.407394);                 // Beijing, China 
//    set_LoRa_Region(-1.292066, 36.821945);                  // Nairobi, Tanzania
    ACTIVE_REGION = LORAMAC_REGION_US915;
    SerialUSB.println(F("LoRa Region:        \tUS902928_NAMERICA_F"));
  };

  // Set the LoRa Frequency based on Region
  switch (ACTIVE_REGION){
    case LORAMAC_REGION_AS923:
      LMIC_setupChannel(0, 923200000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(1, 923400000, DR_RANGE_MAP(0, 6),  1);
      LMIC_setupChannel(2, 923600000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(3, 923800000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(4, 924000000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(5, 924200000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(6, 924400000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(7, 924600000, DR_RANGE_MAP(0, 5),  1);
//      LMIC_setupChannel(8, 924800000, DR_RANGE_MAP(DR_FSK, DR_FSK),   1); 
      break;
    case LORAMAC_REGION_EU868:
      LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(1, 868300000, DR_RANGE_MAP(0, 6),  1);
      LMIC_setupChannel(2, 868500000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(3, 867100000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(4, 867300000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(5, 867500000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(6, 867700000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(7, 867900000, DR_RANGE_MAP(0, 5),  1);
//      LMIC_setupChannel(8, 868800000, DR_RANGE_MAP(DR_FSK,  DR_FSK),  1);
      break;
    case LORAMAC_REGION_US915:
      // NA-US and AU channels 0-71 are configured automatically
      // but only one group of 8 should (a subband) should be active
      // TTN recommends the second sub band, 1 in a zero based count.
      // https://github.com/TheThingsNetwork/gateway-conf/blob/master/US-global_conf.json
      LMIC_selectSubBand(1);
      break;
    case LORAMAC_REGION_AU915:
      // NA-US and AU channels 0-71 are configured automatically
      // but only one group of 8 should (a subband) should be active
      // TTN recommends the second sub band, 1 in a zero based count.
      // https://github.com/TheThingsNetwork/gateway-conf/blob/master/US-global_conf.json
      LMIC_selectSubBand(1);
      break;
    case LORAMAC_REGION_CN779:
      LMIC_setupChannel(0, 779500000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(1, 779700000, DR_RANGE_MAP(0, 6),  1);
      LMIC_setupChannel(2, 779900000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(3, 780500000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(4, 780700000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(5, 780900000, DR_RANGE_MAP(0, 5),  1);
      break;
    case LORAMAC_REGION_IN865:
      LMIC_setupChannel(0, 865062500, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(1, 865402500, DR_RANGE_MAP(0, 6),  1);
      LMIC_setupChannel(2, 865985000, DR_RANGE_MAP(0, 5),  1);
      break;
    case LORAMAC_REGION_RU864:
      LMIC_setupChannel(0, 864100000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(1, 864300000, DR_RANGE_MAP(0, 6),  1);
      LMIC_setupChannel(2, 864500000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(3, 864700000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(4, 864900000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(5, 868900000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(6, 869100000, DR_RANGE_MAP(0, 5),  1);
      break;
    case LORAMAC_REGION_KR920:    
      LMIC_setupChannel(0, 922100000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(1, 922300000, DR_RANGE_MAP(0, 6),  1);
      LMIC_setupChannel(2, 922500000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(3, 922700000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(4, 922900000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(5, 923100000, DR_RANGE_MAP(0, 5),  1);
      LMIC_setupChannel(6, 923300000, DR_RANGE_MAP(0, 5),  1);
      break;
    default:
      // NA-US and AU channels 0-71 are configured automatically
      // but only one group of 8 should (a subband) should be active
      // TTN recommends the second sub band, 1 in a zero based count.
      // https://github.com/TheThingsNetwork/gateway-conf/blob/master/US-global_conf.json
      LMIC_selectSubBand(1);
      break;
  }
  
  // Disable link check validation
  LMIC_setLinkCheckMode(0);

  // TTN uses SF9 for its RX2 window.
  LMIC.dn2Dr = DR_SF9;

  // Set data rate and transmit power for uplink (note: txpow seems to be ignored by the library)
  LMIC_setDrTxpow(DR_SF10, 20);

  // Start job (sending automatically starts OTAA too)
  do_send(&sendjob);
`

**Environment**

Version of LMIC being used: 3.2.0
Version of Arduino IDE being used: Arduino 1.8.13
Network provider (The Things Network, Swisscom, ChirpStack, etc.): TTN
Region (EU868, US915, etc.): US915 but would like to use multiple at the same time
Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.): Custom
Radio (HopeRF, SX1276, etc.): SX1276 from CMWX1ZZABZ-091

terrillmoore, 2020-09-11:
@AndreiHanu when I refactored the code to support multiple regions, I designed things so that it "would be possible in principle" to have dynamic region selection. Meaning that I've never done it.

The biggest issue is that the LMIC code changes the "shape" of `struct LMIC_t` at compile time based on the region selection. There are well-known ways to handle this by using unions in key places, but it would take a few days. It would be rather invasive change, because everywhere in the code where it says "`LMIC.channel[]` would have to become `LMIC.regional.eulike.channel[]`, and so forth. Then everything would need to be retested. I keep thinking that I'll have some time to work on the LMIC soon, but I'm involved in a large USB project which (combined with my non-engineering duties) means I have no time. Sigh. And this is a big enough change that other changes (like the pending join-duty cycle change) would have to be reworked.

That said, obviously it's of interest to me. Just don't know what to say about how to get it done. I think someone with a high level of understanding of the code would need to refactor `lmic.h`, and then possibly others could go find and fix the compile errors, and report back on the mistakes in the refactoring...

AndreiHanu, 2020-09-13:
Thank you Terry!

Are you specifically referring to this piece of code?

`
#if CFG_LMIC_EU_like // EU868 spectrum ====================================================

enum { MAX_CHANNELS = 16 };      //!< Max supported channels
enum { MAX_BANDS    =  4 };

enum { LIMIT_CHANNELS = (1<<4) };   // EU868 will never have more channels
//! \internal
struct band_t {
    u2_t     txcap;     // duty cycle limitation: 1/txcap
    s1_t     txpow;     // maximum TX power
    u1_t     lastchnl;  // last used channel
    ostime_t avail;     // channel is blocked until this time
};
TYPEDEF_xref2band_t; //!< \internal

struct lmic_saved_adr_state_s {
    u4_t        channelFreq[MAX_CHANNELS];
    u2_t        channelMap;
};

#elif CFG_LMIC_US_like  // US915 spectrum =================================================

enum { MAX_XCHANNELS = 2 };      // extra channels in RAM, channels 0-71 are immutable

struct lmic_saved_adr_state_s {
    u2_t        channelMap[(72+MAX_XCHANNELS+15)/16];  // enabled bits
    u2_t        activeChannels125khz;
    u2_t        activeChannels500khz;
};

#endif // ==========================================================================
`

terrillmoore, 2020-09-13:
Yes, that's it.

closed 2020-12-10 by terrillmoore