mcci-catena/arduino-lmic #603: Can't get OTAA to work TTGO OLED v1 TTN 868MHz

jdeus, 2020-09-13:
Hello,

I own a MikroTik LoRa gateway (Switzerland) and I'm trying to make TTN work with OTAA using a TTGO v1 (OLED, ufl on the front, 3D antenna on the back)

- **In ABP mode, I can send a message which is correctly received in the TTN console.**
- The module itself works (tried receiving and sending using the LoRa module in a point-to-point configuration)
- The gateway sends the JoinAccept message
- I tried `LMIC_setClockError(MAX_CLOCK_ERROR * 10 / 100);` but no change.
- Both uplink and downlink are confirmed working on the gateway
- I can send from the TTN console a downlink message, which is received by the ESP32 (tested in ABP mode)

Has anyone been able to get OTAA working with this library ?

Here's my config : 

> const lmic_pinmap lmic_pins = {
    .nss = 18,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 14,
    .dio = {26, 33, 32},
};

> // project-specific definitions
#define CFG_eu868 1
#define CFG_sx1276_radio 1
#define LMIC_DEBUG_LEVEL 2

Log :

> Starting
RXMODE_RSSI
4756: engineUpdate, opmode=0x8
Packet queued
4779: EV_JOINING
4783: engineUpdate, opmode=0xc
141152: engineUpdate, opmode=0xc
141169: EV_TXSTART
141238: TXMODE, freq=868500000, len=23, SF=7, BW=125, CR=4/5, IH=0
455603: setupRx1 txrxFlags 00 --> 01
start single rx: now-rxtime: 4
456235: RXMODE_SINGLE, freq=868500000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 459259 rxtime: 456228 entry-rxtime: 3031 now-entry: 5 rxtime-txend: 311126
517853: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 4
518485: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 534885 rxtime: 518478 entry-rxtime: 16407 now-entry: 4 rxtime-txend: 373376
534902: processRx2Jacc txrxFlags 0x2 --> 00
535642: EV_JOIN_TXCOMPLETE: no JoinAccept
538442: engineUpdate, opmode=0xc

**TTN Device log:**

[image]

**TTN Gateway log:**

[image]

**Lora gateway:**

(Taken a while later)

[image]

terrillmoore, 2020-09-13:
Yes, OTAA works. The TTN console does not indicate the gateway sent the packet; the gateway might not receive the downlinks, and the console only shows what TTN thinks it sent. I've been misled many times by this. You have to look at the gateway logs; perhaps Microtik can help with that.

Most likely there's a misconfiguration on the antenna configuration: your board may have an external switch for TX/RX on the antenna, and the LMIC has to be told how to handle it (it's not handled by the SX1276). Or it could be something else; unfortunately, we don't have the TTGO boards at MCCI.

If you have two units, get the "raw" (low-level device-to-device) example working. If not, it's difficult to bring things up. There may be someone using the TTGO hardware watching this who can give you advice.

jdeus, 2020-09-13:
I've included the gateways logs (last screenshot) : the message is received from TTN and sent over the air. Its downlink is confirmed to be operational.
As stated, I've successfully tested these boards in point-to-point configuration with success (https://github.com/LilyGO/TTGO-LORA32/tree/LilyGO-868-V1.0)
There's no TX/RX pin for that board and as far as I've seen, folks have been able to use the same configuration as I have with success. Then again, lots of people can't get OTAA to work.

JackGruber, 2020-09-16:
Hi, 
I have absoulte no problem with OTAA on 868 Mhz on various projects.

Board: TTGO T3_V1.6
Lmic Version: 3.2.0.0
```
    #define PIN_LMIC_NSS  18
    #define PIN_LMIC_RST  14
    #define PIN_LMIC_DIO0 26
    #define PIN_LMIC_DIO1 33
    #define PIN_LMIC_DIO2 32
```
[image]

cyberman54, 2020-09-19:
Suggestion: To ensure, that your hardware and LORAWAN setup is properly working, you maxy install [this software](https://github.com/cyberman54/ESP32-Paxcounter), which comes with [preconfigured option for TTGOv1](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/platformio.ini#L16). It is based on MCCI LMIC 3.2.0 and proven by many users to work with TTN & OTAA & TTGOv1.

closed 2020-12-10 by terrillmoore