mcci-catena/arduino-lmic #604: Setting Keys Dynamically APPEUI DEVEUI  APPKEY

peter3099, 2020-09-19:
Hello

Firstly thank you for all the hard work on this library it works great.  I'm looking for some guidance on how to implement setting the LoraWan keys dynamically, mainly using a webserver running on the same device so keys don't need to be added to the program code.
 I'm aware of the os_getDevEui,os_getArtEui,os_getDevKey callbacks but I'm getting confused about how would I convert a string that I get through the webserver into a variable that the arduino-lmic library would understand. 

I'm using ESP32.  

Thanks in advance

 Version of LMIC being used: 3.2.0
- Version of Arduino IDE being used:  PlatformIO 
- Network provider ChirpStack
- Region EU868
- Board ESP32
- Radio SX1276

cyberman54, 2020-09-19:
Look e.g. in [this example](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/lorawan.cpp#L153)

terrillmoore, 2020-09-19:
MCCI does everything dynamically -- see https://github.com/mcci-catena/Catena-Arduino-Platform for how we do it. We store the keys in FRAM.  There is a refactor that is currently on a branch: https://github.com/mcci-catena/arduino-lmic/tree/devel-issue578, that refactors the lower edge to create a "secure element" abstraction. It works, but I've not merged it yet because my team is still working on making the Trusted Objects soft secure element work with it, and I anticipate there may be things I overlooked. But if you look at the apis -- you can run doxygen after you've cloned the branch, I use https://scoop.sh to install doxygen on Windows, and it works well -- you'll be able to see the structure somewhat.

An intermediate approach is shown in https://github.com/mcci-catena/Arduino-LoRaWAN, which intercepts the APIs and presents a structured C++ wrapper (without needing you to use all the other MCCI framework).

peter3099, 2020-09-22:
Thank for all the help,  I'm looking into the pax counter example and I think I have what I need, I will close this for now, migth reopen it if I get stuck.

Thanks again

closed 2020-09-22 by peter3099