mcci-catena/arduino-lmic #605: LMIC_setDrTxpow() does not function properly for AS923

e-tinkers, 2020-09-21:
Hi, Thanks and appreciated all the great works in this fork of Arduino-LMIC, I chose to use this version of the fork among all the Arduino-LMIC implementation for its extensive works on supporting all the regional configurations. 

It seems that `LMIC_setDrTxpow()` doesn't function properly for AS923, and would like to seek for your clarification on whether this is by-design or some sort of bug? 

1) setting data rate
I noticed that no matter what data rate I used, the radio is always transmitted at SF9BW125, passing in DR_12 or DR_7 as function argument to `LMIC_setDrTxpow()` does not seem to have any effect on the actual transmission data rate. Although I do noticed that the SF10BW125 is always used for the joining the network, but afterward all the transmissions are at SF9BW125. 

I tried set `LMIC_setAdrMode()` to `true` or `false` with the same result.  If I print out the `LMIC.datarate` on Serial Monitor, the value does change accordingly from 0 to 5 when `SF_12` to `SF_7` is used as the function argument, but the result on TTN always shown that SF9BW125 is used.

2) setting tx power
The print out of `LMIC.Txpow` always shows as 0 no match what value is used when calling `LMIC_setDrTxpow(DR_SF7, x), I understand that according to Lora regional parameters documentation, that the default value for AS923 end node is set to 14dbm. However, unlike EU, my understanding is that it is not a MAX_EIRP in the case of AS923, and therefore should be able to be configurable as long as it not exceeding the local telecom authority's regulated power requirements,  or in some case I should be able to reduce the power to save energy. 

So my question what is the expected behaviour of `LMIC_setDrTxpow()` under AS923? If this function is not working as expected, is there any workaround in the meantime?

**Environment**
- Version of LMIC being used: 3.2.0
- Version of Arduino IDE being used: 1.8.13
- Network provider (The Things Network, Swisscom, ChirpStack, etc.): TTN
- Region (EU868, US915, etc.): AS923
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.): RAK811 ( STM32L1 MCU + SX1276)
- Radio (HopeRF, SX1276, etc.): SX1276

terrillmoore, 2020-09-28:
I've run pre-compliance tests for AS923, which tests tx power and spreading factors; so I'm not certain that it's a bug at all.

However, bear in mind that this version of the LMIC checks your packet size against the regulations. In particular, if it thinks your packet is too large for SF10, it will change to SF to a lower spreading factor to attempt to make it fit. There are _strict variants on the transmit functions that defeat this. Check your messages size against the regional parameters to be sure.

As for transmit power, there are a number of bugs against this. Initially, it didn't work at all. Currently it does have an effect for me (again, this is tested in the compliance tests with the RedwoodCom tester). However, you can't really assume that LMIC.TxPow is what was used. I am still terribly busy with other things, but I think you'll find if you dig through the code that the transmit power is based on ADR and  other downlinks from the network. This version of the LMIC is really aiming for compliance, which is in tension with the device's ability to control what's going on.

I'll keep this on the list to investigate.

e-tinkers, 2020-10-02:
Thanks for the detail explanation and sorry for take a few days to response as I was busy on another project and have to put this one away for a few days.

My payload was 16 bytes when I observed SF9 is used. I did some tests after receiving your reply. The data rate DOES change based on the length of payload (and no matter what parameter is used on `LMIC_setDrTxpow()` function), and I observed that SF7 is used when payload is 200 bytes, SF8  on payload=100 bytes, and so on... However, when the payload is less than 12 bytes, it stay at SF10, even when the payload is reduced to 1 byte, there is no use of SF11 and SF12, is this the correct behaviour that SF11/SF12 will not be used?

On TX power, I read your reply as that it is affected by how the server set the ADR? Right now I don't have enough knowleadge and understanding on the subject yet, I guess that I will have to dig through the source code to have better understanding before I can do further testing. any guiding on where I should look into would be helpful.

terrillmoore, 2020-12-10:
The LMIC will only lower the DR (raise the SF) under very limited circumstances.

- if link integrity checking is enabled, and the LMIC has not seen any downlink traffic after some number up uplinks, it will gradually increase the SF. (The idea is that the device might be on a fringe and increasing power might help hit the gateway.) 
- if the network sends a downlink telling it to do so.

It's bad for the network to use more bandwidth than is needed to hit the gateway. (Remember Kant's categorical imperative: devices should behave in a way that will maximize network capacity if *every* device behaved that way.)  If your traffic is getting through at SF7, you should stay at SF7.

The adaptive procedure outlined above comes from the LoRaWAN spec. Digging through that spec is really required in order to understand the LMIC.

Hope this helps.

closed 2020-12-10 by terrillmoore