mcci-catena/arduino-lmic #607: Documentation has wrong default for LMIC_PRINTF_TO

matthijskooijman, 2020-09-27:
**Describe the bug**

The README says:

> Changing debug output
>        #define LMIC_PRINTF_TO SerialLikeObject
> This variable should be set to the name of a Serial-like object, used for printing messages. If not defined, Serial is assumed.

However, when this is not defined in the project config, `printf`-based printing is not available at all (and enabling debug logging will likely fail or even crash).

See also
https://github.com/mcci-catena/arduino-lmic/blob/89c28c5888338f8fc851851bb64968f2a493462f/src/lmic/config.h#L93
https://github.com/mcci-catena/arduino-lmic/blob/89c28c5888338f8fc851851bb64968f2a493462f/src/hal/hal.cpp#L374-L414

terrillmoore, 2020-09-28:
I will be happy to accept a patch for this; I don't use printf-based debugging at all, after my negative experiences during compliance, but I'm happy to make it work for others. It would be great also to have a regression test (at least a travis-level regression test) -- I imagine it's possible to catch a misconfiguration with `#if`/`#error`.

matthijskooijman, 2020-09-28:
> I imagine it's possible to catch a misconfiguration with #if/#error.

Not really, since you might intentionally not configure `LMIC_PRINTF_TO` if your Arduino core or another library already arranges for `printf` / `stdout` to print somewhere meaningful.

I'm about to create a PR to fix the documentation issue, maybe I'll do some more polishing on the printf debugging separately later (no promises, though).

PR #610 (2020-12-05): Better document printf support
This fixes #607.
commit a6721099 (2020-12-05): Better document printf support

This fixes #607.
closed 2020-12-05 by terrillmoore