mcci-catena/arduino-lmic #620: Resume OTAA session after sleep - downlink broken

javafrog, 2020-11-04:
I'm having issues with resuming a LoRa session after deep sleep. As I read here https://lora-developers.semtech.com/resources/tools/basic-mac/programming-model-and-api/, that can be achieved by saving the configuration as RTC_DATA_ATTR and restoring it via void `LMIC_setSession(u4_t netid, devaddr_t devaddr, u1_t* nwkKey, u1_t* artKey)`

> Set static session parameters. Instead of dynamically establishing a session by joining the network, precomputed session parameters can be provided. To resume a session with precomputed parameters, the frame sequence counters (LMIC.seqnoUp and LMIC.seqnoDn) must be restored to their latest values.

Accordingly, I'm saving netid, devaddr, nwkKey, artKey, seqnoUp and seqnoDn and restoring these after reboot. All of this seems to work perfectly with the regular uplink messages, i.e., they arrive at TTN and are processed there. However, downlink messages only work before I go to sleep for the first time. Afterwards, all downlink messages are reported to be sent by TTN (and also by my gateway), but seem to be ignored by LMIC (0 bytes reported). The problem does not occur if I do not go to deep sleep at all or if I rejoin every time after wake up (which is discouraged).

Is there anything else I need to save and restore?

**Environment**

This information is all important; it's hard to help without a complete set of answers.

- Version of LMIC being used: 3.2.0
- Version of Arduino IDE being used: PlatformIO 5.0.2
- Network provider: TTN
- Region: EU868
- Board: TTGO Lora32 V2.0
- Radio: SX1276

JackGruber, 2020-11-04:
Hi @javafrog when I started I saved the following:
```
LMIC.netid;
LMIC.devaddr;
LMIC.nwkKey
LMIC.artKey
LMIC.dn2Dr;
LMIC.dnConf;
LMIC.seqnoDn;
LMIC.seqnoUp;
LMIC.adrTxPow;
LMIC.datarate;
LMIC.txChnl;    
LMIC.adrAckReq;
LMIC.rx1DrOffset;
LMIC.rxDelay;
LMIC.opmode;
LMIC.channelFreq
LMIC.channelDrMap
LMIC.channelDlFreq
LMIC.bands
LMIC.channelMap;
```
With this a wakeup from deepsleep works fine for up and downlinks. 
The only problem is the loss of the mills and therefore the dutycycle is not correct! 

for this you have to reset the bands
```
for (u1_t bi = 0; bi<MAX_BANDS; bi++) 
{
    LMIC.bands[bi].avail = 0;
 }
```

But now I just save the whole structure.
https://jackgruber.github.io/2020-04-13-ESP32-DeepSleep-and-LoraWAN-OTAA-join/

javafrog, 2020-11-04:
Hi @JackGruber,
thanks, what you are writing seems to be adressing exactly my issue - I'll look into it, thanks! Wish I had found your blog during my Google search ;) I'll let you know if it works out.

javafrog, 2020-11-05:
I just tested it and it works like a charm! Thanks a lot!

closed 2020-11-05 by javafrog