mcci-catena/arduino-lmic #622: Remove required DIO pins

Bartvelp, 2020-11-07:
**Is your feature request related to a problem? Please describe.**
Currently I am using an ESP8266, which has limited GPIO output and would like to shrink the required number of pins to just the MOSI, MISO, CLK and CS pins. (and perhaps RST but I am not sure it is necessary for the RFM95W) As far as I am able to understand, the DIO pins are used for interrupts, which are not used by default, as you need to enable them in the project config file and by reading this issue: https://github.com/mcci-catena/arduino-lmic/issues/332#issuecomment-507373838

**Describe the solution you'd like**
Implement the pull request made to matthijs kooijmans repo by hallard
https://github.com/matthijskooijman/arduino-lmic/pull/34

**Describe alternatives you've considered**
I have considered just removing the asserts by the library but I am not sure if that would lead to unexpected state/behaviour.

**Additional context**
I am not LMIC proficient enough to determine if I could just remake the pull request in this repo by hallard or that too much has changed in the mean time in the under lying implementation.

terrillmoore, 2020-12-10:
Duplicate of #577

terrillmoore, 2020-12-10:
As part of cleaning things up, I'm closing duplicates. Sorry you were having problems!

closed 2020-12-10 by terrillmoore