mcci-catena/arduino-lmic #627: Failure oslmic.c:53

sxwid, 2020-11-18:
I refer to [Issue 561](https://github.com/mcci-catena/arduino-lmic/issues/561). I experience the same. I try to run the ttn-abp demo code on a Arduino UNO with a Dragino LoRa Shield v1.4.
os_init(); is failing with the exact same error. In my understanding it is a problem of wiring. But according to the official wiring here the codesnippet should be ok:

```
const lmic_pinmap lmic_pins = {
  .nss = 10,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = 9,
  .dio = {2, 6, 7},
};
```
Debug Mode Level 2 is not possible due to not enogh memory. Level1 doesn't show interesting stuff. Or I am looking at the wrong place. Is there another approach I can take?

**Environment**
- MCCI LMIC library 3.2.0
- Arduino 1.8.13
- The Things Network
- Region EU868
- Arduino UNO with a Dragino LoRa Shield v1.4

terrillmoore, 2020-11-18:
Check your SPI bus initialization. The lmic_pinmap doesn't include the SPI pins; the library assumes that the MOSI and MISO lines are hooked up appropriately.

Use the raw example (much smaller) and you should be able to turn on debug prints.

manuelbl, 2020-11-18:
I agree with @terrillmoore that it's most likely hardware related. There is one additional possible issue specific to the Dragino LoRa Shield: RST behave differently.

During startup, the LMIC library drives the RST pin high for about 1ms and then puts it in floating state (see https://github.com/mcci-catena/arduino-lmic/blob/master/src/lmic/radio.c#L1101 and https://github.com/mcci-catena/arduino-lmic/blob/master/src/hal/hal.cpp#L68). This is working for all boards except the Dragino shield. Instead of floating, it needs to be driven low. It might or might not apply to your version of the shield.

I've never figured out why this board behaves differently.

sxwid, 2020-11-20:
Thank you for the inputs! After your comments, I remembered that I had issues with the SPI bus before on this Arduino-Clone (WAVGAT...) Using an original Arduino solved the issue. I didn't had to change anything on the RST nor SPI Init. Thanks for the hint!

closed 2020-11-20 by sxwid

alifj98, 2020-12-30:
Hi guys, i am facing the same issue as this and i believe my arduino uno board is original. However, will order another one soon to just be sure. In the meanwhile, it would like to try if there is any other solutions to this problem. As @terrillmoore mentioned, i need to check my SPI bus initialization. How do i do this? Thank you guys if you could help me. 

MCCI LMIC library 3.2.0
Arduino 1.8.13
The Things Network
Region AS923
Arduino UNO with a Dragino LoRa Shield v1.4

also tried with Dragino GPS LoRa Shield v.13